commaai/openpilot #23318: Loading screen while driving

MJR1989, 2021-12-27:
### Describe the bug

Openpilot was driving and suddenly the screen turned upside down for a moment then the screen turned off and the comma loading screen popped up openpilot never stopped driving the car but the video recording also turned off and froze on the uploaded video pulled over unplugged the c2 and plugged it back in and it functioned as designed afterwards 
[image]

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

606ed67f89d79ddf|2021-12-26--23-11-25--3

### openpilot version

0.8.12

pd0wm, 2021-12-28:
@adeebshihadeh Duplicate of https://github.com/commaai/openpilot/issues/21539?

adeebshihadeh, 2022-01-01:
Yes, it's likely a duplicate. @MJR1989 can you connect your device to wifi and make sure to upload all the logs for the route in connect.comma.ai?

adeebshihadeh, 2022-01-13:
Looks like the rlogs are gone by now. This is likely a duplicate of a known issue that we plan to address in the next release.

closed 2022-01-13 by adeebshihadeh