commaai/openpilot #23340: hyundai longitudinal enabled state out of sync with control state

gregjhogan, 2021-12-31:
I noticed the hyundai car controller when using hyundai longitudinal the `enabled` flag can get out of sync with controls state
example: `5decac85ab437ed1|2021-12-24--12-48-45--75`
[image]
Note how we continue to send ACC command for a short period of time after brake press (although the brake controller already disabled on the brake press)

gregjhogan, 2021-12-31:
looking at the code, I didn't understand how this is possible, so I looked in cabana and it seems things do line up

[image]

perhaps this is a plotjuggler issue?

adeebshihadeh, 2021-12-31:
Double check with matplotlib and open an issue on plotjuggler?

gregjhogan, 2021-12-31:
It could be how we load the data for plotjuggler, for example accurate timing requires proper use of the bus time for each CAN message that rolls over frequently, which I know cabana does found [here].(https://github.com/commaai/cabana/blob/368e6df07daa33ecdf9fefa45d867f402dd979a2/src/workers/CanStreamerWorker.worker.js#L31-L59)

gregjhogan, 2021-12-31:
Actually my plot from cabana wasn't even the same bus (128 vs 0) and since the message was sent by openpilot bus 128 makes sense and bus 0 for this message doesn't even show up in cabana.  I am not sure why we have this message on bus 0 since it doesn't show up in cabana (I can't think of a way cabana could filter it out if it was real)

[image]

so it seems like this an issue with something to do with how we process the data for plotjuggler

jyoung8607, 2022-01-01:
> It could be how we load the data for plotjuggler, for example accurate timing requires proper use of the bus time for each CAN message that rolls over frequently, which I know cabana does found [...]

I haven't looked into the underlying code at all, but commaai/cabana#79 may be of interest to you, insofar as busTime (or lack thereof after the recent major Panda refactor) might affect your results in either PlotJuggler or Cabana.

gregjhogan, 2022-01-01:
I see in the log that the messages are on bus 0 and 128 and the bus time is always zero
```
--
    ( address = 1057,
      busTime = 0,
      dat = "\000 \000\xf7\xe3~\000\x8a",
      src = 0 ),
--
    ( address = 1057,
      busTime = 0,
      dat = "\000 \000\xf7\xe3~\000\x8a",
      src = 128 ),
--
```

gregjhogan, 2022-01-01:
bus 0 messages for this address only seem to be from `sendcan` (this probably explains why they do not show up in cabana) so not sure why they show up at all under `can` not to mention with different timing.

here is the data from the log for `can` and `sendcan` filtered by addr 1057 and the time slice in the above screenshot (notice no transition between 56.0 and 56.1 seconds)
```
55.603024 bus=128 addr=0x421 data=002000bb625700f5 source=can
55.614478 bus=  0 addr=0x421 data=002000ab62550016 source=sendcan
55.621317 bus=128 addr=0x421 data=002000ab62550016 source=can
55.633868 bus=  0 addr=0x421 data=002000ab62550007 source=sendcan
55.642895 bus=128 addr=0x421 data=002000ab62550007 source=can
55.655103 bus=  0 addr=0x421 data=002000ab625500f8 source=sendcan
55.661959 bus=128 addr=0x421 data=002000ab625500f8 source=can
55.674516 bus=  0 addr=0x421 data=002000a1225400d9 source=sendcan
55.681444 bus=128 addr=0x421 data=002000a1225400d9 source=can
55.695227 bus=  0 addr=0x421 data=002000a1225400ca source=sendcan
55.702867 bus=128 addr=0x421 data=002000a1225400ca source=can
55.714826 bus=  0 addr=0x421 data=002000a1225400bb source=sendcan
55.722013 bus=128 addr=0x421 data=002000a1225400bb source=can
55.733941 bus=  0 addr=0x421 data=000000ffe37f00fc source=sendcan
55.741721 bus=128 addr=0x421 data=000000ffe37f00fc source=can
55.755641 bus=  0 addr=0x421 data=000000ffe37f00ed source=sendcan
55.762383 bus=128 addr=0x421 data=000000ffe37f00ed source=can
55.774198 bus=  0 addr=0x421 data=000000ffe37f00de source=sendcan
55.782004 bus=128 addr=0x421 data=000000ffe37f00de source=can
55.794491 bus=  0 addr=0x421 data=000000ffe37f00b0 source=sendcan
55.802587 bus=128 addr=0x421 data=000000ffe37f00b0 source=can
55.814081 bus=  0 addr=0x421 data=000000ffe37f00a1 source=sendcan
55.822130 bus=128 addr=0x421 data=000000ffe37f00a1 source=can
55.833281 bus=  0 addr=0x421 data=000000ffe37f0092 source=sendcan
55.842555 bus=128 addr=0x421 data=000000ffe37f0092 source=can
55.855456 bus=  0 addr=0x421 data=000000ffe37f0083 source=sendcan
55.861767 bus=128 addr=0x421 data=000000ffe37f0083 source=can
55.874868 bus=  0 addr=0x421 data=000000ffe37f0074 source=sendcan
55.881948 bus=128 addr=0x421 data=000000ffe37f0074 source=can
55.893623 bus=  0 addr=0x421 data=000000ffe37f0065 source=sendcan
55.902992 bus=128 addr=0x421 data=000000ffe37f0065 source=can
55.914895 bus=  0 addr=0x421 data=000000ffe37f0056 source=sendcan
55.921388 bus=128 addr=0x421 data=000000ffe37f0056 source=can
55.934119 bus=  0 addr=0x421 data=000000ffe37f0047 source=sendcan
55.942243 bus=128 addr=0x421 data=000000ffe37f0047 source=can
55.955379 bus=  0 addr=0x421 data=000000ffe37f0038 source=sendcan
55.961892 bus=128 addr=0x421 data=000000ffe37f0038 source=can
55.974387 bus=  0 addr=0x421 data=000000ffe37f0029 source=sendcan
55.981184 bus=128 addr=0x421 data=000000ffe37f0029 source=can
55.995517 bus=  0 addr=0x421 data=000000ffe37f001a source=sendcan
56.003003 bus=128 addr=0x421 data=000000ffe37f001a source=can
56.014389 bus=  0 addr=0x421 data=000000ffe37f000b source=sendcan
56.022240 bus=128 addr=0x421 data=000000ffe37f000b source=can
56.033800 bus=  0 addr=0x421 data=000000ffe37f00fc source=sendcan
56.041640 bus=128 addr=0x421 data=000000ffe37f00fc source=can
56.055636 bus=  0 addr=0x421 data=000000ffe37f00ed source=sendcan
56.062508 bus=128 addr=0x421 data=000000ffe37f00ed source=can
56.073903 bus=  0 addr=0x421 data=000000ffe37f00de source=sendcan
56.081997 bus=128 addr=0x421 data=000000ffe37f00de source=can
56.094858 bus=  0 addr=0x421 data=000000ffe37f00b0 source=sendcan
```

sshane, 2022-02-17:
Resolved by https://github.com/commaai/PlotJuggler/pull/40

closed 2022-02-17 by sshane