commaai/openpilot #23342: CI: reduce LFS

adeebshihadeh, 2021-12-31:
Need to check if we can fit all of the LFS files in the Actions cache. also don't pull LFS files unless used

L3Fingers, 2021-12-31:
I apologize, did not realize what I clicked would affect something besides on my side. Was not trying to pull or do anything.

tecandrew, 2022-01-01:
going to test on for x86 mac, but failed to clone/checkout due to LFS-related issue. didn't explicitly state to pull LFS items, but threw an error anyway

```bash
╭─atec@trabus-mba in ~/workspace/op
╰$ git clone https://github.com/commaai/openpilot.git --recurse-submodules
Cloning into 'openpilot'...
remote: Enumerating objects: 59486, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 59486 (delta 0), reused 0 (delta 0), pack-reused 59485
Receiving objects: 100% (59486/59486), 1.64 GiB | 25.80 MiB/s, done.
Resolving deltas: 100% (34029/34029), done.
Updating files: 100% (1897/1897), done.
Downloading models/dmonitoring_model.onnx (3.3 MB)
Error downloading object: models/dmonitoring_model.onnx (895ee32): Smudge error: Error downloading models/dmonitoring_model.onnx (895ee32e2a1c77496e015270db475eef65034b25331f2859bac0ccf702f64298): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

Errors logged to /Users/atec/workspace/op/openpilot/.git/lfs/logs/20211231T191133.165091.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: models/dmonitoring_model.onnx: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

adeebshihadeh, 2022-01-01:
@tecandrew we spent all our LFS budget this month. It'll reset tomorrow.

rzeydelis, 2022-01-03:
I think the budget has reset, but I am getting this error when trying to `git lfs pull`:

Error updating the git index: (5/5), 131 MB | 24 MB/s                                                
error: common/__init__.py: cannot add to the index - missing --add option?
fatal: Unable to process path common/__init__.py

closed 2022-01-03 by adeebshihadeh