commaai/openpilot #23343: poor GPS reception

antaod, 2021-12-31:
### Describe the bug

I just got C3. After driving for a few minutes, an error message pop-up "poor GPS reception". It occurred every trip 

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

a96d2488318ec4b8|2021-12-31--02-15-22--0

### openpilot version

0.8.12

### Additional info

position of camera
[image]
[image]

L3Fingers, 2021-12-31:
This is not something I reported. I don't believe that's my car. I haven't submitted any bug or trouble tickets for my C3. Unless its something the C3 did itself?
Thank you,John

robbederks, 2022-01-03:
I've taken a quick look at the qlogs, and it looks like there are no raw gps messages coming in, waiting for your device to come online to confirm.
Most likely the cable between your main PCB and the gps module might have come undone in shipping. If you're comfortable disassembling the device you can take a look yourself to see if both sides of the cable are fully seated, if not, please contact support@comma.ai and we'll have your device fixed in house! Sorry for the trouble!

antaod, 2022-01-03:
> Most likely the cable between your main PCB and the gps module might have come undone in shipping. If you're comfortable disassembling the device you can take a look yourself to see if both sides of the cable are fully seated, if not, please contact ***@***.*** and we'll have your device fixed in house! Sorry for the trouble!

Thank you for checking this for me.

Antao

closed 2022-01-03 by adeebshihadeh