commaai/openpilot #23397: Subaru: immediate disengagements from EPS fault

adeebshihadeh, 2022-01-04:
How confident are we in this signal? We had 14 immediate disengagements due to this on 0.8.12.

Are we sure it's a permanent fault? If it's only temporary, we can move this to `steerWarning` which will make it a less harsh soft disable.

```
522ff91c7c405658|2021-12-16--20-46-31 5 'SUBARU FORESTER 2019'
522ff91c7c405658|2021-12-18--00-00-14 6 'SUBARU FORESTER 2019'
68bab2ef27edf65b|2021-12-19--14-55-31 30 'SUBARU FORESTER 2019'
68bab2ef27edf65b|2021-12-19--18-39-15 22 'SUBARU FORESTER 2019'
522ff91c7c405658|2021-12-20--03-23-04 42 'SUBARU FORESTER 2019'
522ff91c7c405658|2021-12-20--08-24-48 25 'SUBARU FORESTER 2019'
9c6b37e7e356b1e4|2021-12-21--20-41-10 12 'SUBARU FORESTER 2019'
68bab2ef27edf65b|2021-12-21--15-30-49 13 'SUBARU FORESTER 2019'
68bab2ef27edf65b|2021-12-21--15-45-46 13 'SUBARU FORESTER 2019'
68bab2ef27edf65b|2021-12-21--17-54-47 35 'SUBARU FORESTER 2019'
2a1043a0db0fffb4|2021-12-26--19-40-45 16 'SUBARU IMPREZA LIMITED 2019'
68bab2ef27edf65b|2021-12-28--19-10-18 52 'SUBARU FORESTER 2019'
68bab2ef27edf65b|2022-01-03--15-39-03 15 'SUBARU FORESTER 2019'
68bab2ef27edf65b|2022-01-03--15-55-25 11 'SUBARU FORESTER 2019'
```

adeebshihadeh, 2022-01-04:
@martinl do you know anything about this? Let me know if you want the above logs

martinl, 2022-01-05:
["Steering_Torque"]["Steer_Error_1"] is set when eyesight hard faults and car needs restart. I usually get it during testing when I stop openpilot > eyesight faults > start openpilot without restarting the car. I have also seen it when max torque is exceeded for Impreza/Crosstrek 2020+ but not with Forester. I can take a look at he logs.

closed 2022-07-17 by adeebshihadeh