commaai/openpilot #23398: Chrysler: immediate disengagements from EPS fault

adeebshihadeh, 2022-01-04:
How confident are we in this signal? We had 15 immediate disengagements due to this on 0.8.12.

Are we sure it's a permanent fault? If it's only temporary, we can move this to steerWarning which will make it a less harsh soft disable.

```
625623dea0c4dc1b|2021-12-16--17-07-21 25 'CHRYSLER PACIFICA HYBRID 2018'
fc5df38c8fcae904|2021-12-16--17-47-12 18 'CHRYSLER PACIFICA HYBRID 2018'
fc5df38c8fcae904|2021-12-17--10-58-27 28 'CHRYSLER PACIFICA HYBRID 2018'
91d1e9944d9dc0eb|2021-12-19--11-03-17 5 'CHRYSLER PACIFICA HYBRID 2018'
6586d1e6b127683b|2021-12-19--12-37-55 11 'CHRYSLER PACIFICA HYBRID 2019'
fc5df38c8fcae904|2021-12-20--10-34-36 48 'CHRYSLER PACIFICA HYBRID 2018'
fc5df38c8fcae904|2021-12-20--14-14-56 19 'CHRYSLER PACIFICA HYBRID 2018'
91d1e9944d9dc0eb|2021-12-20--14-48-52 4 'CHRYSLER PACIFICA HYBRID 2018'
26a085ad528b1d93|2021-12-22--15-55-08 30 'CHRYSLER PACIFICA 2018'
c071afb883596e77|2021-12-24--18-53-01 4 'CHRYSLER PACIFICA 2018'
5e7e194a940e5e60|2021-12-28--19-02-34 5 'CHRYSLER PACIFICA HYBRID 2017'
26a085ad528b1d93|2021-12-30--12-52-15 8 'CHRYSLER PACIFICA 2018'
1f77c729c55939cc|2021-12-30--18-36-08 12 'JEEP GRAND CHEROKEE V6 2018'
0a3e89f78b1d0071|2021-12-31--09-58-44 11 'CHRYSLER PACIFICA 2018'
4bcff61eb824ed6c|2022-01-02--15-33-27 88 'CHRYSLER PACIFICA HYBRID 2018'
```

adeebshihadeh, 2022-01-04:
@adhintz do you know anything about this? Let me know if you want the above logs

adhintz, 2022-01-05:
Feel free to share the logs with me and I'll take a look. pushespretn@gmail.com or drew on discord.

The only EPS fault I knotw of is having OP engaged at a stop, then holding down the accel pedal while crossing over min steer speed.

adeebshihadeh, 2022-01-06:
@adhintz None of them have rlogs uploaded yet. We're writing something to request them while the device is offline. I'll update you in this thread once they're available.

adhintz, 2022-01-06:
I'm suspicious of [this code](https://github.com/commaai/openpilot/blob/cf9792dc536db343447cfa64edecc72565c4eefd/selfdrive/car/interfaces.py#L154):
```
    # Disable on rising edge of gas or brake. Also disable on brake when speed > 0.
    # Optionally allow to press gas at zero speed to resume.
    # e.g. Chrysler does not spam the resume button yet, so resuming with gas is handy. FIXME!
    if (cs_out.gasPressed and (not self.CS.out.gasPressed) and cs_out.vEgo > gas_resume_speed) or \
       (cs_out.brakePressed and (not self.CS.out.brakePressed or not cs_out.standstill)):
      events.add(EventName.pedalPressed)
```
What might happen is gasPressed goes to True while vEgo <= gas_resume_speed. This means it does not add the pedalPressed event. However later when vEgo > gas_resume_speed the previous gasPressed state (stored in self.CS.out.gasPressed) is True so it does not add the pedalPress event because it only wants to add the pedalPressed even on the rising edge.

With no pedalPressed event, OP/ACC is never canceled.

Then the fault occurs because the [panda safety code notices that gas is pressed](https://github.com/commaai/panda/blob/ca67da626fa0bb71f31adcde0662614b3fe59828/board/safety/safety_chrysler.h#L103), the car is going fast enough, but that OP is still engaged. So the safety code sets [controls_allowed = 0](https://github.com/commaai/panda/blob/801fd4997d35f81232e95e033f95afa46fa6e3fc/board/safety.h#L206), dropping transmit messages, causing an EPS fault.

However maybe this suspicion is wrong because the faults on 2017 occur around 3.8m/s (min steer speed), not around 2m/s (gas_resume_speed).

adhintz, 2022-01-08:
Just went for a drive and this change does fix the EPS fault I described: https://github.com/adhintz/openpilot/commit/15dcc8486bda90067af8f156a6135b248cbc1b48

adhintz, 2022-01-11:
Any thoughts on getting rid of the rising edge condition [from here](https://github.com/commaai/openpilot/blob/cf9792dc536db343447cfa64edecc72565c4eefd/selfdrive/car/interfaces.py#L154)? It doesn't seem to be needed, but I also don't know why it was added in the first place.

Perhaps something like this:
```
    # Disable on gas or brake. Disable on brake when speed > 0.
    # Optionally allow to press gas at zero speed to resume.
    # e.g. Chrysler does not spam the resume button yet, so resuming with gas is handy. FIXME!
    if (cs_out.gasPressed and (cs_out.vEgo > gas_resume_speed)) or \
       (cs_out.brakePressed and not cs_out.standstill):
      events.add(EventName.pedalPressed)
```

If you're fine getting rid of the rising edge condition, please let me know -- I'll test and send a PR.

jyoung8607, 2022-01-12:
Also see #20724

adhintz, 2022-01-12:
> Also see #20724

Thanks! That's the same EPS fault condition that's fixed with https://github.com/adhintz/openpilot/commit/15dcc8486bda90067af8f156a6135b248cbc1b48

jyoung8607, 2022-01-12:
Pretty sure we need that. Gas rising edge is to handle the case of the driver engaging at a steady cruise with their foot still on the gas, which is a pretty common usage. openpilot goes into preEnabled immediately, then enabled when gas is released.

Rising edge of brake is to handle engaging at a stop with a lead car with the driver's foot on the brake, OP can take it from there when the driver releases the brake.

I think I agree there's opportunities for peril here, in that openpilot and Panda look at the gas state a little differently, but I would need to study a logged event.

Maybe related: I have a note from another Discord user "itemus" that they take a LKAS fault if they try to go in-gear too quickly after startup, a second or two. I haven't talked to him enough to see if that's truly a gear thing (feels unlikely) and maybe an in-motion or gas-press situation at the moment of safety switch and controlsd car port startup (feels more likely).

jyoung8607, 2022-01-12:
> However maybe this suspicion is wrong because the faults on 2017 occur around 3.8m/s (min steer speed), not around 2m/s (gas_resume_speed).

I think you're on the right track. Panda can indeed go controls_allowed = 0 at ~2m/s, but it's still perfectly happy to pass zero torque steering messages from openpilot, which engaged or not, [is exactly what gets sent](https://github.com/commaai/openpilot/blob/85dcfbbf360705fc40b048cb256158d89c7dcf98/selfdrive/car/chrysler/carcontroller.py#L38-L41) below 3.8m/s for the 2017. So, no Panda discards below that point, and therefore no EPS fault.

jyoung8607, 2022-01-12:
Consider tweaking this: https://github.com/commaai/openpilot/blob/85dcfbbf360705fc40b048cb256158d89c7dcf98/selfdrive/car/chrysler/carcontroller.py#L38

to this:
```python
lkas_active = moving_fast and enabled and not CS.out.gasPressed
```

When engaging at cruising speeds, controlsd does this for you in preEnable when gas is still held, keeps the actuators at zero until the driver releases the gas. You'll note Panda has nothing special for preEnable, that's what the rising edge detection is for.

The driver should only be giving a momentary tap to signal ACC to resume, so no steer for that near-standstill moment shouldn't be problematic. In a perfect world, controlsd would know more about gas resume, would hold actuators at zero for the duration, but so long as Chrysler is the only car that needs it, may as well fix it with three words in the Chrysler port.

adeebshihadeh, 2022-01-12:
> Then the fault occurs because the panda safety code notices that gas is pressed, the car is going fast enough, but that OP is still engaged. So the safety code sets controls_allowed = 0, dropping transmit messages, causing an EPS fault.

Unfortunately, all the rlogs are gone by now. On the bright side, we now have it setup to automatically upload the rlogs once these events are flagged. I checked the qlogs for a mismatch between `pandaStates[0].controlsAllowed` and `controlsState.enabled`, and it appears you're right that panda was just blocking the messages. Then, the car faults before openpilot's controlsMismatch shows up.

adeebshihadeh, 2022-01-12:
> ```python
> lkas_active = moving_fast and enabled and not CS.out.gasPressed
> ```

This is on the right track, but enabled should be replaced with active. Enabled is active or preEnabled.

jyoung8607, 2022-01-12:
> Why do we check the speed in the first place?

Expedient hax, or so I would assume. I think a proper solution would be teaching generic_rx_checks() to do rising edge checks against `last_gas_pressed || standstill` or something like that. Then we stop caring how late they release.

Before someone spends time on Panda test cases for that, it would be good to know if Chrysler resume spam is truly impossible, or just something that needs to be fixed. @realfast has been doing some work in this area and may have insight.

> This is on the right track, but enabled should be replaced with active. Enabled is active or preEnabled.

Hey, two bugfixes for the price of one!

jyoung8607, 2022-01-12:
> ```
> 625623dea0c4dc1b|2021-12-16--17-07-21 25 'CHRYSLER PACIFICA HYBRID 2018'
> fc5df38c8fcae904|2021-12-16--17-47-12 18 'CHRYSLER PACIFICA HYBRID 2018'
> fc5df38c8fcae904|2021-12-17--10-58-27 28 [...]

Worth noting all but one of these are from the 3.8 m/s (less than 5mph) minSteerSpeed group where it's much easier to accidentally trip this on ACC resume. There's only one from the 17.5 m/s (38 mph) minSteerSpeed group.

adeebshihadeh, 2022-01-14:
Thanks @jyoung8607 and @adhintz!

closed 2022-01-14 by adeebshihadeh