commaai/openpilot #23401: locationd: improve device falling detector

adeebshihadeh, 2022-01-04:
10/15 events while engaged on 0.8.12 were false positives

```
49708c324cfe37cb|2021-12-17--21-40-48--8
49708c324cfe37cb|2021-12-18--21-39-10--8
49708c324cfe37cb|2021-12-18--21-53-11--0
9d5eed56285661da|2021-12-19--15-14-02--42
4aa808e86ad0b42c|2021-12-23--09-44-43--13
9d5eed56285661da|2021-12-25--04-49-30--21
9d5eed56285661da|2021-12-25--04-49-30--25
fec5a5fe860afaa2|2021-12-25--10-05-51--26
fec5a5fe860afaa2|2021-12-25--10-05-51--28
9d5eed56285661da|2021-12-25--16-06-00--13
8a50d9d15a073bd2|2021-12-26--16-36-37--6
887e76d536c1841e|2021-12-28--08-26-52--15
4ee121621f388597|2021-12-28--16-44-15--74
0445433fbd48dfc9|2021-12-30--14-52-47--150
b91277ed62ccc40a|2022-01-01--23-46-02--16
8baf540cff3444d0|2022-01-03--06-48-54--27
```

adeebshihadeh, 2022-01-06:
Disabled for now

HaraldSchafer, 2022-11-16:
No good balance between FP and FN found

closed 2022-11-16 by HaraldSchafer