commaai/openpilot #23403: Error while running scons -j16

rzeydelis, 2022-01-05:
### Describe the bug

I am continuing my journey with following George's [video](https://www.youtube.com/watch?v=ixfAdv9sL30) on running openpilot on my Ubuntu machine. I am not sure this is the right place for my noob questions, but I do appreciate the help. So far I have done the following:
1. Cloned the repo onto my machine
2. Made sure I am using Python 3.8
3. Ran `./ubuntu_setup.sh`
4. Ran `git lfs pull`

When I run `scons -j16`, I get the following error:
```
scons: Reading SConscript files ...

scons: warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'cereal/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 367, in <module>

scons: warning: Ignoring missing SConscript 'rednose/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 399, in <module>

scons: warning: Ignoring missing SConscript 'cereal/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 403, in <module>

scons: warning: Ignoring missing SConscript 'panda/board/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 404, in <module>

scons: warning: Ignoring missing SConscript 'opendbc/can/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 405, in <module>

scons: *** Import of non-existent variable ''libkf''
File "/home/bobby/projects/op_venv/openpilot/selfdrive/locationd/SConscript", line 1, in <module>
```
I noticed the error is mentioning that I am missing SConscript in the cereal folder. When I check my cereal folder, I noticed it is empty. Am I missing a command I am supposed to be running?

### OS Version

Ubuntu 20.4

rzeydelis, 2022-01-05:
Was able to fix the cereal issue by removing the empty cereal folder and running this command:
`git clone https://github.com/commaai/cereal.git`

Now after re-running the `scons -j16` command, I am getting this error:

```
scons: Reading SConscript files ...

scons: warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'rednose/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 399, in <module>

scons: warning: Ignoring missing SConscript 'panda/board/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 404, in <module>

scons: warning: Ignoring missing SConscript 'opendbc/can/SConscript'
File "/home/bobby/projects/op_venv/openpilot/SConstruct", line 405, in <module>

scons: *** Import of non-existent variable ''libkf''
File "/home/bobby/projects/op_venv/openpilot/selfdrive/locationd/SConscript", line 1, in <module>
```

I figured I could fix it the same way, but it seems like rednose isn't a directory?

whowe444, 2022-01-05:
You need the virtual env. Follow the readme in openpilot/tools. Basically clone with recursive submodules, run Ubuntu setup, activate pipenv, and then you can build.

adeebshihadeh, 2022-01-05:
You're missing the submodules. The `--recurse-submodules` is supposed to take care of this, but it seems that it can silently fail. I updated the README to do an explicit `git submodule update --init`.

closed 2022-01-05 by adeebshihadeh