commaai/openpilot #23424: Comma 3 XC - master-ci Random Reboot

Memberoffoxhound, 2022-01-06:
### Describe the bug

Comma 3 XC
-Master CI 0.8.13

Started experiencing random reboots yesterday evening after re-enabling DM recording. I was parked and idling when the events occurred. 

Something of interest is that assuming the name of the route includes the date, the route following the first reboot had a September date. I'm not sure if. that's of any significance other than it certainly didn't take place then. 

If anything else is needed, I'm happy to oblige. Thanks!

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

ff0e4e4791c2ccf5|2022-01-04--18-41-50--0

### openpilot version

0.8.13

### Additional info

Affected route examples
ff0e4e4791c2ccf5|2022-01-04--18-41-50--0

ff0e4e4791c2ccf5|2021-09-07--13-38-17--0 <-This is the route/segment immediately following the one above. I'm unsure why it's labeled 2021-09-07

pd0wm, 2022-01-06:
Looked at the logs and one of the power regulators is getting hotter than normal. This probably causes some instabilities in voltage, which then causes the flash to temporarily stop operating ultimately leading to the reboot.

Contact support@comma.ai so your device can be replaced.

Memberoffoxhound, 2022-01-06:
Thanks much @pd0wm 

@VirtuallyChris sorry for the double email to support just now... Apple mail is goofy sometimes.

closed 2022-01-06 by pd0wm