commaai/openpilot #23444: Random reboot while on-road

Hubblesphere, 2022-01-07:
### Describe the bug

After driving 30 minutes C3 rebooted causing dash errors with no warning. Looking at footage the segment ends about 1 minute before the reboot happened so footage is missing from the end of the drive. Reboot happened just as I was getting to my destination so device went off road shortly after it happened. 

openpilot version 0.8.12 release3
device: comma 3
Vehicle: 2021 Toyota Highlander
Last available segment: 8d8427a2116eace2|2022-01-06--18-19-23--2

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

8d8427a2116eace2|2022-01-06--18-19-23--2

### openpilot version

0.8.12 release3

### Additional info

Left device overnight in vehicle, drove next day with no issues so after going off-road it seems to be back to normal.

pd0wm, 2022-01-07:
Can you upload rlogs?

Hubblesphere, 2022-01-07:
It shows rlogs for the last partial segment 3 as "none."

pd0wm, 2022-01-07:
Requested segment three manually for you. Since the qlog is corrupted you can't do it through connect.

pd0wm, 2022-01-07:
From the logs/next bootlogs I don't see anything immediately obvious. From the PMIC status it looks like the device rebooted out of it's own will, indicating a false positive on the power drop monitor. When the battery voltage drops below 7V we turn off the screen, try to flush as much as possible to disk and shut down cleanly to prevent disk corruption. https://github.com/commaai/agnos-builder/blob/master/userspace/usr/comma/power_drop_monitor.py

The power drop monitor tries to write a param when this happens (`LastPowerDropDetected`), but this wasn't set in your case. Did some more testing on my own device, and this doesn't seem as reliable as I thought. So the absence of the param doesn't really prove anything.

Let me know if this happens to you more often! If it does happen you can try replacing your cable.

In the meantime I'll try to improve the logging around the power drop monitor, and ensure we can better debug this.

Hubblesphere, 2022-01-07:
Just for context, I recently changed the USB-C cable due to a random connection dropout issue but not a power issue. This new cable has been working for a few months without issue, but could be a similar cable failing issue. If so these C3 cables are only lasting about 3 months at a time. And I'm leaving device and cable connected as much as possible so no excessive wear from handling. 

I'll report if it happens again, and swap cables again to see if that resolves it. 

Thanks for looking into it!

pd0wm, 2022-01-07:
In the next bootlogs there are no `pstore` entries (logs from the last boot held in RAM). When I manually run the `halt -f` that the power drop monitor runs to shut down, the `pstore` entries remain. This could indicate your device actually lost power briefly.

alancyau, 2022-01-07:
@Hubblesphere Possible there are any kinks the way you're routing the USB cable? Would be weird to see the cable fail twice. Just throwing some ideas around.

Hubblesphere, 2022-01-08:
@alancyau first one failed on my 2019 Corolla and I swapped to the new one after getting my 2021 Highlander. They have different installation positions and I think Highlander gives a bit more room for the cable. If it does turn out to be the cable again I'll try to reassess my installation and avoid any tight bends.

alancyau, 2022-01-19:
I believe my C3 also experienced a random reboot while the car was turned on. I was sitting in the back seat, so I didn't catch the entire reboot process. If I recall, OP was disengaged and the car was being driven manually at the time. The driver noticed the C3 screen go dark and saw the Comma symbol. I believe a reboot occured.

pd0wm, 2022-01-19:
@alancyau What is your dongle id, and approximate time of the reboot?

alancyau, 2022-01-19:
@pd0wm The reboot occurred on 1/14/22 during a 7 hour road trip. I don't see the route on Comma Connect. Do you think it never got uploaded? I believe my device is still in Prime trial mode. If I can see the route in connect, I can skim through the drive and estimate when the reboot occured.

Dongle ID: a4af1602d8e668ac

pd0wm, 2022-01-19:
Looking at the logs that was a ui crash, which restarts automatically. Happened 1/14 on 20:09 UTC. Don't see any bootlogs from that day, so your device itself did not crash.

alancyau, 2022-01-19:
Thanks for checking the logs. Would a UI crash cause OP to disengage if OP was engaged prior to the UI crash?

pd0wm, 2022-01-19:
No, it would stay engaged. It would throw a `processNotRunning` alert, which is a no entry condition (i.e. doesn't allow you to enter the engaged state).

adeebshihadeh, 2022-08-14:
@Hubblesphere has this happened again? If not, I think we can close this, and if it's happened again, we should have some better logging now.

Hubblesphere, 2022-08-14:
No issues since.

closed 2022-08-14 by Hubblesphere