commaai/openpilot #23458: apple plotjuggler doesn't run on fresh pull & build of OP

tecandrew, 2022-01-09:
### Describe the bug

plotjuggler `--install` doesn't run on first attempt (fresh pull, build of OP, and run of `juggle.py` script). the app works if building and running as standalone `./plotjuggler` from the fork repo

```
((openpilot) )  ~/workspaces/openpilot/tools/plotjuggler/ [master] ./juggle.py --install
Traceback (most recent call last):
  File "/Users/lafcadio/workspaces/openpilot/tools/lib/logreader.py", line 9, in <module>
    from xx.chffr.lib.filereader import FileReader
ModuleNotFoundError: No module named 'xx'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./juggle.py", line 14, in <module>
    from selfdrive.test.process_replay.compare_logs import save_log
  File "/Users/lafcadio/workspaces/openpilot/selfdrive/test/process_replay/compare_logs.py", line 15, in <module>
    from tools.lib.logreader import LogReader
  File "/Users/lafcadio/workspaces/openpilot/tools/lib/logreader.py", line 11, in <module>
    from tools.lib.filereader import FileReader
  File "/Users/lafcadio/workspaces/openpilot/tools/lib/filereader.py", line 1, in <module>
    from tools.lib.url_file import URLFile
  File "/Users/lafcadio/workspaces/openpilot/tools/lib/url_file.py", line 8, in <module>
    import pycurl
ImportError: pycurl: libcurl link-time ssl backends (secure-transport, openssl) do not include compile-time ssl backend (none/other)
```

### OS Version

macOS m1 monterey

### openpilot version or commit

a0eed0cdd23598fd8dcc34805cc83ed0fe515348

adeebshihadeh, 2022-08-14:
Still not working for you? It's working on my machine on master.

tecandrew, 2022-08-14:
can try in a sec. it's been a minute

tecandrew, 2022-08-14:
`mac_setup.sh` doesn't work at the moment. fails on installation of `ffmpeg`

```
==> Installing ffmpeg dependency: libsodium
==> Pouring libsodium--1.0.18_1.arm64_monterey.bottle.tar.gz
🍺  /opt/homebrew/Cellar/libsodium/1.0.18_1: 73 files, 779.7KB
==> Installing ffmpeg dependency: zeromq
==> Pouring zeromq--4.3.4.arm64_monterey.bottle.tar.gz
🍺  /opt/homebrew/Cellar/zeromq/4.3.4: 83 files, 6.0MB
==> Installing ffmpeg dependency: zimg
==> Pouring zimg--3.0.4.arm64_monterey.bottle.tar.gz
🍺  /opt/homebrew/Cellar/zimg/3.0.4: 27 files, 1MB
==> Installing ffmpeg
==> Pouring ffmpeg--5.1.arm64_monterey.bottle.tar.gz
🍺  /opt/homebrew/Cellar/ffmpeg/5.1: 282 files, 49.3MB
==> Running `brew cleanup ffmpeg`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
Installing ffmpeg has failed!
Installing glfw
Installing libarchive
Installing libusb
Installing libtool
Installing llvm
Installing openssl
Installing pyenv
Installing qt@5
Installing zeromq
Installing protobuf
Installing protobuf-c
Installing swig
Homebrew Bundle failed! 1 Brewfile dependency failed to install.
lafcadio@atec-mbp openpilot/tools (master) »
```

tecandrew, 2022-08-14:
doesn't work still. rerunning the setup script results in success and was able to build using `scons`. although I don't think it installed `ffmpeg`.

running PJ results in this error:

```
((openpilot) ) lafcadio@atec-mbp tools/plotjuggler (master) » ./juggle.py "4cf7a6ad03080c90|2021-09-29--13-46-36"                                     
PlotJuggler is out of date. Installing update...
PlotJuggler-3: Unknown option 'window_title'.
```

adeebshihadeh, 2023-06-21:
macOS support should be in a good state now. Feel free to open new issues if you encounter any

closed 2023-06-21 by adeebshihadeh