commaai/openpilot #23467: C3 Ui Crash w/ OP Engaged

jsegill, 2022-01-09:
### Describe the bug

OP was enabled and performing fine. Suddenly the Toyota dash cluster showed the "grab the wheel" notification and the C3 made a soft warning sound but no warning message was displayed on the C3. Then the C3 showed the comma symbol, and after ~15 seconds returned to a running state. OP released control during the ui crash and OP only regained control when I pressed the cruise control buttons. 

Attached is a conversation w/ Willem @pd0wm about the issue in Discord. https://discord.com/channels/469524606043160576/616456819027607567/928263624928145438

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

99725b56a15b5510|2022-01-04--14-35-13--8

### openpilot version

0.8.12

pd0wm, 2022-01-10:
Looking at the logs looks like the GPU reset/froze:
```
[557.859469] MAIN 0 kernel - kgsl kgsl-3d0: _ui[46016]: gpu timeout ctx 7 ts 24603
[557.859616] MAIN 0 kernel - kgsl kgsl-3d0: _ui[46016]: gpu fault ctx 7 ts 24603 status 00E71105 rb 15f0/16e0 ib1 00000007F85AE000/0ead ib2 0000000000000000/0000
[557.859694] MAIN 0 kernel - kgsl kgsl-3d0: _ui[46016]: gpu fault rb 2 rb sw r/w 15f0/16e0
[557.859762] MAIN 0 kernel - kgsl kgsl-3d0: |a6xx_snapshot_gmu| set FENCE to ALLOW mode:0
[557.859825] MAIN 0 kernel - kgsl: kgsl_snapshot_push_object: snapshot: Can't find entry for 0x0000000000000000
[557.859882] MAIN 0 kernel - kgsl: adreno_snapshot: GPU snapshot froze 804Kb of GPU buffers
[557.859941] MAIN 0 kernel - kgsl kgsl-3d0: |kgsl_device_snapshot| snapshot created at pa 0x000000016d200000 size 871764
[557.860258] MAIN 0 kernel - kgsl kgsl-3d0: |a6xx_reset| Device soft reset failed
[557.860330] MAIN 0 kernel - platform 506a000.qcom,gmu: Suspended GMU
[557.860427] MAIN 0 kernel - kgsl kgsl-3d0: _ui[46016]: gpu failed ctx 7 ts 24603
```

This then triggered a ui bug, that is now fixed (https://github.com/commaai/openpilot/commit/8d80c0107f5d1cd2ba2b5a37f297d3ae8e3b02f3#diff-4ed470dd9510d089320647133f2fc9a5dd69798a449b5fa9e10dbaab83aa8b85).

pd0wm, 2022-01-28:
Another one in `0f88deb1f9e2c2a1|2022-01-21--11-35-23--2`

pd0wm, 2022-01-31:
`99725b56a15b5510|2022-01-04--14-35-13--8`

jsegill, 2022-02-04:
Worth sending the device back to support for analysis?

pd0wm, 2022-02-06:
Only if it keeps happening. Did it happen again after your initial report?

jsegill, 2022-02-16:
Hasn't happened again to my knowledge since the initial report.

adeebshihadeh, 2022-03-21:
Another instance: `fd1f4eb7549d49fe|2022-03-09--19-57-10--40`

adeebshihadeh, 2022-03-22:
Briefly looked into this while doing #23932. Easy to repro by lowering `/sys/class/kgsl/kgsl-3d0/dispatch/drawobj_timeout`, but not sure why it happens yet.

jsegill, 2022-03-30:
> Another instance: `fd1f4eb7549d49fe|2022-03-09--19-57-10--40`

I didn't get a notification of that while driving...

pd0wm, 2022-03-31:
These other instances we're posting are from other users. We're using this issue to track the general problem.

jsegill, 2022-03-31:
> 

Thanks for clarifying!

adeebshihadeh, 2023-12-08:
Hasn't popped up in a while

closed 2023-12-08 by adeebshihadeh