commaai/openpilot #23484: Joystick 🕹️

ryleymcc, 2022-01-10:
### Describe the bug

Joystickd.py running on PC does not connect to comma device. Instructions should be improved since everyone I asked was unable to get it to work.

### OS Version

Ubuntu 20.04

adeebshihadeh, 2022-01-10:
Let me know if you still have issues after 7eca4d62078b14de62c5f2ce08e382eb31cc7076

closed 2022-01-10 by adeebshihadeh

tgnabor-Phx999, 2022-01-10:
Please unsubscribe me from this thread, I tried- no can do

•Tom Nabor•

ryleymcc, 2022-01-12:
> Let me know if you still have issues after [7eca4d6](https://github.com/commaai/openpilot/commit/7eca4d62078b14de62c5f2ce08e382eb31cc7076)

joystickd.py works with keyboard when running it directly on the c2 using the command `tools/joystick/joystickd.py --keyboard`.

When I do
`echo -n "1" > /data/params/d/JoystickDebugMode`

then,

`cereal/messaging/bridge {192.168.57.189} testJoystick`

then on my laptop 
`export ZMQ=1`

then
`tools/joystick/joystickd.py`

It starts on my laptop but there is no output from the comma 2 ssh shell. It does enter joystick mode but it seems like it's not connected. How can I tell if my laptop is sending the data over the network?