commaai/openpilot #23487: 2022 Ridgeline auto disengages for no reason.

J3ffrey2020, 2022-01-10:
### Describe the bug

Openpilot disengages when car ahead SLIGHTLY slows down. If the car ahead brakes hard openpilot works fine.  

### What hardware does this issue affect?

comma two

### Which car does this affect?

2022 Honda Ridgeline

### Provide a route where the issue occurs

9df3afe7a752c468|2022-01-09--09-52-16

### openpilot version

latest

### Additional info

This happened 2 times in the route I've provided. It did it 2 more times yesterday.. please let me know if you need anything else.

J3ffrey2020, 2022-01-11:
*No pedal, 100% stock software and hardware.

pd0wm, 2022-01-11:
Can you list which disengagements (with timestamp) in the route where unintended? There are 9 of them.

J3ffrey2020, 2022-01-11:
> Can you list which disengagements (with timestamp) in the route where unintended? There are 9 of them

Sorry! Should have said that, the one at 10:27 am est on Sunday 9th is a great example of one. I would need to provide other routes for the other examples.

pd0wm, 2022-01-11:
It seems like the stock system is trying to AEB and takes over because it applies more braking that openpilot, which then forces a disengagement.

Are things better if you disable stock AEB before plugging in your device? Are there any sensitivity settings you can change?

pd0wm, 2022-01-11:
Once we confirmed that AEB is actually the problem we can figure out a way to make it work together with openpilot. Requiring it to be disabled is not a solution.

J3ffrey2020, 2022-01-11:
> Once we confirmed that AEB is actually the problem we can figure out a way to make it work together with openpilot. Requiring it to be disabled is not a solution.

Excellent, it will probably be a few days before I can really test it out, wont be on the road much till Friday.

J3ffrey2020, 2022-01-17:
Okay I got some fresh data for you. Long story short, even with Emergency Braking turned off it still shows the problem.

I have some examples of it working correctly and then in the same route it fails to work correctly twice.

This is the route:
9df3afe7a752c468 | 2022-01-17--12-01-25

At 12:04 it fails to stop the car, instead just disengaging. 

At 12:09 it also fails to stop 

around 12:13 it works as expected and slows the car down

at 12:19, it also works as expected, slows the car down, and almost stops the car, it disconnects open pilot at somewhere between 2 and 4 mph.

This was all with EAB turned completely off. 

Let me know if I can do anything else to help? It's pretty much 50/50 on if the car stops like it should or if OP disengages. 

The one thing I notice different now, is that with EAB turned off there's no red lines where it disconnects, does that mean there's no error? Even if it's still showing the problem.

pd0wm, 2022-01-20:
Thanks for the info! It was showing the red lines as an indication that stock AEB was triggering/preparing, so not an error.

It seems like two fields in the BRAKE_COMMAND message that we set to zero for other Honda Nidec cars are not actually zero. Specifically:

 - `SET_ME_X00` is actually 8 on your car
 - `SET_ME_X00_2` seems to go through different states based on braking

Can you record a drive in dashcam mode (turn off the openpilot toggle in settings) with some highway driving behind a car and some full stopping? Then I can try to figure out how `SET_ME_X00_2` behaves. Maybe if we set that field correctly it won't disengage.

J3ffrey2020, 2022-01-20:
Will do! Will have some data for you by the end of the weekend! Thank you for your assistance. What is this X00_2 doing ?

J3ffrey2020, 2022-01-22:
Okay, I have some good data for you from this morning. The first Drive I did not have AEB on, the 2nd and 3rd I did. 

During this drive: 9df3afe7a752c468|2022-01-22--10-17-29--0
At 10:27 I first encountered a car. at 10:29 and 10 seconds the car pulls off to the left of the road and makes a pretty quick stop, normally I would have intervened however I wanted to get some good data so I let the car do it on it's own, including the dash board screaming at me to hit the brake, but as far as I can tell AEB was disabled so it didn't do anything other than slow down because of the regular braking system.

The next drive to focus on is this one: 9df3afe7a752c468|2022-01-22--11-00-12--0
There will be a ton of data here, sometimes I have to hit the brakes manually, sometimes I let the car do it and it doesn't cut out till 12ish, even though it's supposed to give up on anything under 25.. at 11:07 and 50 seconds a car makes a left exit, which should also get some good data. During this stop I noticed a lot of clicking going on under the dash, near where you rest your left foot. This sounds like the car is doing something else vs when it's normally braking. This clicking is electronic and NOT the ABS building pressure. The reason I bring this up is because I have heard this clicking noise right before OP gives up in the past.. not sure if it has anything else to do with it, but it just seems like the car is "thinking" more about what it's doing in this kind of situation. 

Either way that should be enough data to see what X00 2 is doing... I have to turn OP back on because the stock honda steering is just awful!! 

Thanks for your help!

J3ffrey2020, 2022-01-27:
Any chance you see what's wrong ?

> Thanks for the info! It was showing the red lines as an indication that stock AEB was triggering/preparing, so not an error.
>
> It seems like two fields in the BRAKE_COMMAND message that we set to [...]

pd0wm, 2022-01-27:
This seems to be the original source of that line: https://github.com/commaai/openpilot/pull/475

Your brake pedal has a bit of noise, causing it to false trigger I think. After eye balling on of your drives, raising the threshold to 0.1 should be enough.

Installing using `installer.comma.ai/commaai/pilot-brake-threshold` should install the correct branch.

[image]

J3ffrey2020, 2022-01-27:
Awesome, thank you. I do remember seeing something about brake noise. Would getting honda to replace the pedal fix do anything or for some reason the honda brake controllers with nidac are no good?

pd0wm, 2022-01-27:
Think your new 2022 model has just a bit more noise than the previous years. Increasing the threshold a little bit will probably solve your issues.

J3ffrey2020, 2022-01-27:
Thank you sir, I'll get it installed asap!

J3ffrey2020, 2022-01-29:
> Think your new 2022 model has just a bit more noise than the previous years. Increasing the threshold a little bit will probably solve your issues.

First drive went great. Is there an official build that has these modifications + works with a pedal?

closed 2022-01-30 by pd0wm

sammy1859, 2022-02-14:
As of today  openpilot show support up to 2021 ridgeline. Are these mods included in a new build? Need to know before I buy. Thanks

PR #23641 (2022-01-30): Honda Pilot, Passport, Ridgeline: increase user brake threshold
resolves #23487 

original PR: https://github.com/commaai/openpilot/pull/475