commaai/openpilot #23488: Black Screen On Start Up + Error Alert While OP Engaged

JamesL623, 2022-01-10:
### Describe the bug

My car (2022 Camry Hyrbid) was sitting overnight (12ish hours) in about 20-degree weather. Upon turning on the car, my Comma Two had a black screen, but OP was working normally.

At one point while it was engaged, OP threw an error noise, but I couldn't see what it was due to the screen being black. Here is the route at where it occurred: 69567f65ae707320|2022-01-10--15-12-06--10 . It felt like the car started to accelerate before I hit the break and the system disengaged, but not 100% sure if that happened.

Unplugging and replugging the USBC into the Comma Two fixed the black screen.

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

69567f65ae707320|2022-01-10--15-12-06--10

### openpilot version

0.8.12

pd0wm, 2022-01-11:
Looks like you engaged behind a stopped car. When traffic started moving you disengaged using the accelerator, and then engaged openpilot again, which then triggered a forward collision warning since it thought there was a risk with collision of the lead car.

Weird that the screen stayed off. First time that happened?

JamesL623, 2022-01-11:
@pd0wm Got it.

Yes, the first time it has happened on my C2. I have only owned it for about 2 months now, but never had the issue with my original EON

adeebshihadeh, 2022-01-12:
Looks like the display power was set correctly according to the logs. If this happens again, contact support to swap out your device.

```
2022-01-10 07:04:35.273520 422 422 [SurfaceFlinger] Set power mode=2, type=0 flinger=0x7fae7c3c00
2022-01-10 07:04:35.273791 422 422 [SDM] HWCDisplay::SetPowerMode: display = 0, mode = 2
2022-01-10 07:04:35.273845 422 422 [SDM] DisplayBase::SetDisplayState: Set state = 1, display 0
```

closed 2022-01-12 by adeebshihadeh

JamesL623, 2022-02-02:
C2 Screen off again: 69567f65ae707320|2022-02-02--00-33-49--1