commaai/openpilot #23540: Looping at "  registering device - serial: xxxxxx , IEMI:(xxxxx, ) on comma 3

shahed2137, 2022-01-15:
### Describe the bug

After resetting comma 3 , its stuck in loop with the msg  'registering device - Serial no xxxxxx , IEMI:(xxxxx, )'

1. Reset comma 3,
2. Connect wifi.
3. Downloading and installation completed 100%
4. Device reboot with Comma logo
5. Stays on Registering device.
6. Stuck at " registering device - Serial no xxxxxx , IEMI:(xxxxx, )" for hours.

Same result while trying in the car or at home. 

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

c78c1972adfa10df|2022-01-14--16-55-00

### openpilot version

0.8.12

### Additional info

1 new install
2. In Melbourne Australia

redacid95, 2022-01-15:
Try reinstalling OP again. This happen to me. It took 2 reinstalls it went away. After that try flashing AGNOS.

adeebshihadeh, 2022-01-17:
What's your serial number? The device needs to be connected to the internet to register. Can you uninstall and make sure you're connected to the internet until it's registered?

shahed2137, 2022-01-17:
Thank you for response.
Before the installation, it ask to connect to wifi and installation starts after it connects to internet.
After the installation, comma 3 reboots automatically and try to register.

I do not know if it remembers the wifi connection and there is no option to reconnect. 

Please see the attached screenshot for serial no.

Regards

Shahed

BrianWisniewski, 2022-01-17:
@shahed2137 The screenshot you attached is not displayed on the issue, likely because you replied via email. Try to upload your image using the GitHub website.

shahed2137, 2022-01-17:
Serial no 5e083c32

Regards

Shahed

shahed2137, 2022-01-18:
Here is a video 

https://1drv.ms/v/s!ApQM42kw4zUGh45YrmzFHeM7wIlQqw

pd0wm, 2022-01-18:
Looks like the request at our server is lacking the public key. However, the token is signed with the correct private key (based on the public key tied to that serial/dongle). So it seem like only your public key on the device is empty (`/persist/comma/id_rsa.pub`). Did you change the contents of those files for some reason?

shahed2137, 2022-01-18:
I am not very technical.  I have entered the system .. 

Any idea how to fix?

Regards

Shahed

shahed2137, 2022-01-18:
I didn’t enable ssh on this device .

Just had my first drive after installing.

Regards

Shahed 

> On 18 Jan 2022, at 9:40 pm, Shahed Jalal ***@***.***> wrote:
>

adeebshihadeh, 2022-01-19:
Support is taking care of this. This shouldn't happen on new devices as the persist partition these keys are written to is now read only. I've also written some more tests to verify that the key pair is valid.

closed 2022-01-19 by adeebshihadeh

shahed2137, 2022-01-19:
> Support is taking care of this. This shouldn't happen on new devices as the persist partition these keys are written to is now read only. I've also written some more tests to verify that the key pair is valid.

Last email from Chris from Comma support .  I have not yet received the fix:

"

Chris McCammon (comma.ai)Jan 17, 2022, 22:26 MSTHi Shahed, We may be able to fix it with software. I’ll send you what to do tomorrow.Best,Chris
--
"

shahed2137, 2022-01-20:
Should I try AGNOS reflash? As support will send software fix,,should I wait?

shahed2137, 2022-01-20:
response from comma support : Please SSH into your device and run these commands.

sudo mount -o rw,remount /persist
openssl rsa -in /persist/comma/id_rsa -pubout > /persist/comma/id_rsa.pub

How do I ssh to comma without network connection? console? any document of video?

shahed2137, 2022-01-20:
I think i am getting a response to ssh requests after the boot before installing openpilot. But failing due to incorrect public key.

I downloaded the openpilot public key from https://github.com/commaai/openpilot/blob/master/tools/ssh/id_rsa

I believe it was posted by you @adeebshihadeh . 

I saved it as id_rsa.ppk and used the setting " private key file for authentication on port 8022 on putty getting attached error

[image]

adeebshihadeh, 2022-01-20:
The username is `comma`, not `root`. Use our [Discord](https://discord.comma.ai) for further help.

shahed2137, 2022-01-20:
I am on discord and posting there too😀

QQ:
I can connect to wifi and pong to comma. But I have no way to enable ssh and load key.

1.Should I be able to ssh without enabling the ssh on device 
2. What authentication method it will use?

shahed2137, 2022-01-20:
I believe I will need private key or password for comma to get in .

shahed2137, 2022-01-22:
I could ssh and run the commands .. the issue of device registering resolved.

However once I connect to my VW Tiguan 2020 drivers assistance.. on the screen it shows 

[image]dashcam mode
Car unrecognised

shahed2137, 2022-01-22:
[image]

EthanChou860314, 2024-03-08:
I uninstall openpilot by heading into Settings -> Software, scrolling to the bottom and tapping "uninstall"
 And I install a custom fork using the comma installer URL "openpilot.comma.ai".
when installing,my internet disconnect, and power is disconnect.

When I power on, my device got stuck on "Registering Device" screen.
I disconnect the power, and then connect power. It's still stuck on "Registering Device" screen.
How can I reset my device?

Now, I use USB-A to USB-C to power on CX3.

EthanChou860314, 2024-03-08:
> ### Describe the bug
> After resetting comma 3 , its stuck in loop with the msg 'registering device - Serial no xxxxxx , IEMI:(xxxxx, )'
>
> 1. Reset comma 3,
> 2. Connect wifi.
> 3. Downloading and [...]

What step can make C3X come back to home page, when stuck on "Registering Device" screen.

EthanChou860314, 2024-04-26:
Factory reset process
It's OK