commaai/openpilot #23568: Car Unrecognized Error

Lhsgenx40, 2022-01-18:
### Describe the bug

I just replaced my front windshield and reconnected my comma 2 v0.8.12 for my 2021 Prius Prime and it started working in the beginning, but then I noticed I had the power cord hanging out the side panel and wanted it to  look better by coming out the hole from the rear view mirror. When I did that I ended up disconnecting the ethernet cord not knowing it disconnect and after that my display continues to say car unrecognized. I rebooted it, uninstalled the software and reinstalled it, reconnected everything and still get the same error.

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

e8dd6bf379531625

### openpilot version

0.8.12

### Additional info

Not sure what provide a route means so I just wrote down my Dongle ID.

adeebshihadeh, 2022-01-18:
This means the comma power isn't properly connected. Ensure that it's fully seated in the OBD port and both ends of the ethernet cable are fully seated. Check out [discord.comma.ai](https://discord.comma.ai) for help.

closed 2022-01-18 by adeebshihadeh

Lhsgenx40, 2022-01-18:
I checked the connections and they are both fully connected. I just checked again and get the same error.