commaai/openpilot #23575: Joystick mode not working

javisaez95, 2022-01-19:
### Describe the bug

I've been trying to use the joystick mode in a 2019 Toyota prius hybrid. For that purpose I'm using a black panda and a toyota car harness. I flashed the black panda firmware to be in debug mode to set controls_allowed to True always. I'm not able to engage openpilot in joystick mode unless I have the cruise mode on and working. The cruise mode only works when the car is above 25 mph, but I want to engage as well when the car is stopped. Is it possible?

Furthermore, when it is engaged (being above 25 mph) and the screen shows: "Joystick mode: Steer X% Gas X%" the car does not respond to the orders I sent to it, but the values of the Steer and gas percentage change on the screen. It only lasts engaged for a few seconds before it starts appearing another messabe saying "Brake! Risk of Collision.".

### OS Version

Ubuntu 20.04

### openpilot version or commit

0.8.10

adeebshihadeh, 2022-01-19:
Check out our [Discord](https://discord.comma.ai) or GitHub Discussions for help.

closed 2022-01-19 by adeebshihadeh

javisaez95, 2022-01-20:
I have checked the Discord server out and the GitHub Discussions, but I did not find any useful information.