commaai/openpilot #23588: Disengage on gas toggle

adeebshihadeh, 2022-01-20:
- [ ] add toggle
- [ ] update safety model
- [ ] panda safety tests
- [ ] ensure smooth transition after gas is released
- [ ] read back unsafe mode
- [ ] match behavior of other systems

pd0wm, 2022-01-20:
if we decide to keep steering while the gas is pressed, I think we should also change our pre-enable state to also start steering immediately.

geohot, 2022-01-21:
I think we definitely want to keep steering while gas is pressed. Super Cruise and Autopilot do.

pd0wm, 2022-01-21:
Yeah, that's what I thought. But currently we don't steer when you engage with the gas pressed. Which could be confusing, since there are two situations where you can press the gas, and only one of them steers. Therefore, I suggest we remove that and also steer when you engage with the gas pressed.

adeebshihadeh, 2022-01-21:
> Yeah, that's what I thought. But currently we don't steer when you engage with the gas pressed. Which could be confusing, since there are two situations where you can press the gas, and only one of them steers. Therefore, I suggest we remove that and also steer when you engage with the gas pressed.

+1 regardless of this situation. `controlsState.active` vs `controlsState.enabled` is super confusing and has led to a bunch of bugs  in the car controllers.

pd0wm, 2022-01-21:
I'm not sure we can get rid of the state completely, since if you have disengage on gas enabled you still want to be able to engage with gas. 

But we should able to remove the active vs enabled thing, and just make something generic based gas pressed. The car controller should not need to be aware of the toggle status, only the state machine logic.

pd0wm, 2022-01-21:
Other TODOs:

- [ ] Ensure the PID controller doesn't wind up when the gas is pressed
- [ ] Ensure we send the correct state for cars with stock longitudinal. Otherwise they might try to maintain zero acceleration, instead of slow down when you press the gas lightly. (https://github.com/commaai/openpilot/issues/22439)

sshane, 2022-01-22:
Another issue I've been trying to solve is that when you override with gas and let go after a large acceleration, the planner starts at aEgo and then winds to what it actually wants to request. So if you accelerate into a lead, it takes quite a while until we start slowing down (we actually accelerate further after gas press).

Just setting accel to 0 while not active works, but on Toyota if you engage at under 10 mph or so, the car always brakes for a second before accelerating. We can try setting the jerk and a_change costs to zero for 1 frame on engagement for that however.

pd0wm, 2022-01-24:
Sounds like setting to zero might be best then. If the Toyota is not following the commands we send it under 10mph, then we need to fix that quirk in the carController, not upstream in in the planner.

zorrobyte, 2022-02-03:
Safety doc update to accompany this PR:
https://github.com/commaai/openpilot/pull/23696

cydia2020, 2022-02-03:
> Another issue I've been trying to solve is that when you override with gas and let go after a large acceleration, the planner starts at aEgo and then winds to what it actually wants to request. So [...]

We can set `PERMIT_BRAKING` to 0 briefly (like for 1 second) if we're sending 0 accel under 10mph, so pcm would just coast along without applying any brakes

pd0wm, 2022-02-03:
@cydia2020 that sounds like a good solution. 

Conditions would be
 - Shortly after engage
 - a_target is positive
 - speed under 10mph

Do you want to make a PR and see if that works?

sshane, 2022-02-03:
Alternatively we can lower/set to zero the jerk costs in the MPC for one frame, so the planner starts at around the acceleration it would in a few seconds. That way we can start accelerating immediately like stock does. It might affect braking though, so testing needs to be done

pd0wm, 2022-02-03:
You mean this? https://github.com/commaai/openpilot/pull/23639

sshane, 2022-02-03:
Yeah it might be better to do this in controls so it's vehicle specific. However I'll gather data on what stock ACC requests when it wants to brake or accelerate when engaging for a few makes.

Maybe they all request their fully ramped up acceleration immediately and can handle that, I'll check that out.

sshane, 2022-02-06:
It definitely seems like Toyota doesn't ramp from 0 much more often than I thought, here's another example of me engaging at a stop. openpilot ACC brakes uncomfortably and then slowly accelerates, while stock ACC requests 0.57 m/s/s immediately: https://imgur.com/a/pDXkVn1

And this is what stock ACC does when you engage behind a car: https://imgur.com/a/bDvCgYf

sshane, 2022-02-11:
Hyundai ACC when it wants to brake, but being overridden by gas press on a Elantra Hybrid:
https://user-images.githubusercontent.com/25857203/153552887-455294ae-b2d6-4fae-8b99-13e18345a584.png

KyleGreenlaw, 2022-03-03:
> Hyundai ACC when it wants to brake, but being overridden by gas press on a Elantra Hybrid. Requested acceleration ramps back to zero at a rate of exactly 0.5 m/s^3 (`(-.32 - -.03) / (75.19 - 75.77)=0.500`):
> https://user-images.githubusercontent.com/25857203/153552887-455294ae-b2d6-4fae-8b99-13e18345a584.png

This also occurs on the Hyundai Sonata Hybrid, the Hyundai Pre-enable fork seemed to fix this if I recall

closed 2022-04-05 by sshane