commaai/openpilot #23601: Visual alert for unintended stock ACC disengagement

Chaheeto, 2022-01-21:
### Describe the bug

"Take Control Immediately" alert doesn't display when the stock ACC disengages automatically on traction control or wheel slip detection.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

72c8dc9054796c61|2022-01-17--06-14-03--32

### openpilot version

0.8.12

### Additional info

Toyota Prius Prime, 2019.

adeebshihadeh, 2022-01-21:
This is the case for all pcmCruise cars right now. Perhaps we should switch to button enable everything on the openpilot side.

pd0wm, 2022-02-28:
As far as I'm aware most Toyota's don't broadcast any button state on CAN. We have to find the signal that indicates this disengagements and alert on them.

adeebshihadeh, 2022-06-12:
#24721

j0dan, 2022-12-27:
So is the issue that the Toyota is disengaging and OP has no way of knowing why?

It feels like a serious safety issue. Having your vehicle start drifting out of lane while driving is a bit freaky.

sshane, 2023-05-10:
https://github.com/commaai/openpilot/issues/23601