commaai/openpilot #23605: Getting "Unknown type name 'AVHWDeviceType'" when trying to compile on pc

Giotje, 2022-01-22:
### Describe the bug

Whenever i try to compile the software i get the "Error: unknown type name 'AVHWDeviceType'.

I'm running linux mint, which i know is not officially supported but i was wondering if i can install something or change something to make it work on Linux mint 20.2

### OS Version

Linux mint 20.2

### openpilot version or commit

Latest

GregorKikelj, 2022-01-22:
Why not just use openpilot docker image?

adeebshihadeh, 2022-01-22:
Unfortunately we really only support Ubuntu 20.04. Perhaps it's worth trying to install everything from `tools/ubuntu_setup.sh`, since Mint is based on Ubuntu. If that doesn't work, @GregorKikelj's suggestion of the docker is a good idea too.

closed 2022-01-22 by adeebshihadeh

Giotje, 2022-01-22:
Hey adeeb and gregor, 

thank you so much for the reply.

i've installed ubuntu 20.04 and it's working butterly smooth.

I hope u have a good rest of your weekend !

adeebshihadeh, 2022-01-22:
Glad to hear the Ubuntu experience was smooth!

deanlee, 2022-01-23:
@Giotje  I work on linux mint most of the time, you may need to run `tools/ubuntu_setup.sh` to install the required libraries.