commaai/openpilot #23606: Unable to open webcam on Openpilot on PC

ramkunchur, 2022-01-22:
### Describe the bug

Hi All,

I am trying to run openpilot with webcam on PC (Ubuntu 20.04)

I've followed all steps, webcam version is successfully built

I've ensured correct webcam IDs are set under below file
selfdrive/camerad/cameras/camera_webcam.cc

Running using following command:
~/Downloads/openpilot/selfdrive/manager$ PASSIVE=1 NOSENSOR=1 USE_WEBCAM=1 ./manager.py

Openpilot opens, click on agreement, app opens, however road side camera doesn't start...

Under settings when I click on "Preview" under Driver Camera, I can see webcam starting...

I am unable to find anything wrong or anything missing...

Any idea why this may not be working?
I don't have Panda, hence using PASSIVE=1

Below is message I get in terminal:
navd ui soundd deleter logmessaged pandad thermald statsd
navd ui soundd deleter logmessaged pandad thermald statsd
navd ui soundd deleter logmessaged pandad thermald statsd...

Appreciate your help and thanks in advance! 

### OS Version

Ubuntu 20.04

### openpilot version or commit

latest

### Additional info

None

pd0wm, 2022-01-25:
Ignition needs be detected high, which you need a panda for.

Since this is a question an not an issue, this is better discussed at discord.comma.ai.

closed 2022-01-25 by pd0wm

daffigusti, 2022-04-13:
I've the same problem, any workaround ?

thanks.