commaai/openpilot #23631: replay error

xiaomanbusiy, 2022-01-26:
### Describe the bug

when i use replay,  **Error sending a packet for decoding**  **camera[0] failed to get frame:###**  happens, how can  i solve this, thanks.

ps: I use command just as below : ./replay '3533c53bb29502d1|2019-12-10--01-13-27--0' --data_dir "../openpilot/data" --no-cuda

### OS Version

Ubuntu 20.04

deanlee, 2022-01-26:
I think the c library is more clear and straightforward than other libraries in this case. there are less than 200 lines of code in consoleui responsible for displaying the interface, the usage pattern of other libraries is very similar to C (set attr, move to position, print, close attr).  they basically don't make the code smaller or easier to read.
I'll cleanup the current code to make it easier to read.

deanlee, 2022-01-26:
Is your video file corrupted? can you play it in the media player?

xiaomanbusiy, 2022-01-26:
>fcamera.hevc can be played well.

xiaomanbusiy, 2022-01-26:
> Is your video file corrupted? can you play it in the media player?

fcamera.hevc can be played well.

deanlee, 2022-01-26:
Have you tried other routes?

xiaomanbusiy, 2022-01-26:
> Have you tried other routes?

I can only download this route data. can you share me other routes data? 
thanks very much.

deanlee, 2022-01-26:
you can replay the demo route with "replay --demo" .

xiaomanbusiy, 2022-01-26:
> you can replay the demo route with "replay --demo" 

My computer is unable to connect to the Internet, so i can only download route data and then replay from local.
so please share me some routes data, thank you very much!

adeebshihadeh, 2022-01-27:
@deanlee should we add a unit test for local files?

deanlee, 2022-01-28:
@xiaomanbusiy    I don't have this problem when replaying local files. I suspect your video file is corrupted. can u pull https://github.com/commaai/openpilot/pull/23649 and run it  to see what error message will come up?

deanlee, 2022-01-28:
> @deanlee should we add a unit test for local files?

added. https://github.com/commaai/openpilot/pull/23650

xiaomanbusiy, 2022-01-28:
> @xiaomanbusiy I don't have this problem when replaying local files. I suspect your video file is corrupted. can u pull #23649 and run it to see what error message will come up?
> 
> Is it convenient for u to upload your fcamera.hevc for testing?

Thanks for your help！
I have merged  #23649,  and tryed, my log is just as below. 
[image]

deanlee, 2022-01-28:
the -1094995529 is FFERRTAG( 'I','N','D','A')  ///< Invalid data found when processing input
It should be that your video file is corrupted

closed 2022-01-28 by adeebshihadeh

xiaomanbusiy, 2022-01-29:
> the -1094995529 is FFERRTAG( 'I','N','D','A') ///< Invalid data found when processing input It should be that your video file is corrupted

ok, thanks very much.
 can you share me your testing data?

pbassut, 2024-11-13:
I have the same problem. The video is uploaded through a comma3x device and downloaded by Cabana. If there's anything corrupted it's in one of those steps: comma device upload of cabana download.