commaai/openpilot #23660: Modulate brakes for idle creep at low speeds, stop and go, instead of harsh state machine

zorrobyte, 2022-01-29:
I've never really liked the long state machine for starting, stopping:
https://github.com/commaai/openpilot/blame/master/selfdrive/controls/lib/longcontrol.py#L31

Humans don't stop completely, then press gas to go again when in creeping at low speeds, like a traffic jam, stop. They modulate the brake pedal and allow engine idle creep advance the car. EVs even emulate this car behavior.

What if we interp brake based on lead distance in the stopping/stopped state `v_ego < 2.0` so the vehicle creeps, then releases back to MPC per usual `v_ego > 2.0`?

zorrobyte, 2022-01-29:
Notes to self, as I may try writing a PR later.
`lead.dRel` seems to be distance, in meters from front bumper to lead. So we'll need to write an interp function between 0-100% brakes based on dRel under 2 v_ego. The state machine then just switches to this interp function then back. Will need to test real world, and possibly on a few different cars.

pd0wm, 2022-01-31:
vEgoStopping is actually 0.5 m/s on most cars. Usually the cars don't respond well to commands we're sending at these low speeds, which is why we have this state machine. Another problem is that wheel speed measurements become unreliable at low speeds. We actually talked about using the localizer at low speeds to improve stop/resume behavior.

Using the distance to the lead car sounds like a hack, and won't work with more e2e stuff. 

I'm going to close this issue since it's not very actionable right now, but we're thinking about ways to improve low sped behavior.

closed 2022-01-31 by pd0wm