commaai/openpilot #23664: Error trying to use Carla

maykonpacheco, 2022-01-29:
### Describe the bug

When I try to run CARLA using the ./start_openpilot_docker.sh command, I always get the same error.
This error does not prevent the UI from opening, but when I finish the entire installation, CARLA does not work.

[image]

[image]

How to reproduce the error.
Open a terminal and run the command
cd tools/sim ./start_carla.sh

Open another terminal and run the command
cd tools/sim ./start_openpilot_docker.sh 

[image]

### OS Version

Ubuntu 20.04

### openpilot version or commit

5fb9c6aa683e8a4ecd7591ecc76c9f29e30b060e

### Additional info

And the first time I'm running openpilot on my computer.

rzeydelis, 2022-01-30:
I am getting the same error. It seems like the "Intro" module was added where you have to go through the tutorial and terms and conditions, which might be leading to this error.

adeebshihadeh, 2022-01-30:
The bridge crash was fixed with 36a3ec04bec98f6b61ada90cdda692dc6fdc0416.

closed 2022-01-30 by adeebshihadeh

maykonpacheco, 2022-01-30:
Now it's working