commaai/openpilot #23674: disengage failure on gas

tecandrew, 2022-01-31:
### Describe the bug

pressing gas did not disable cruise control / openpilot. 

[image]

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

eb5a50f9e98f50b9|2022-01-28--10-48-41--5

### openpilot version

0.8.12

### Additional info

has happened a few times before, but wasn't certain that this was the behavior I experienced.

with the recent development to _not_ disengage on gas, this might be a moot bug report, but submitting it for record's sake.

sshane, 2022-02-04:
Your car in that route isn't using openpilot longitudinal control, as the param `DisableRadar` (a pre-req for longitudinal control) isn't set.

You can see your car params here: https://useradmin.comma.ai/logs?url=https%3A%2F%2Fcommadata2.blob.core.windows.net%2Fqlog%2Feb5a50f9e98f50b9%2F2022-01-28--10-48-41%2F0%2Fqlog.bz2%3Fse%3D2022-02-04T19%253A57%253A04Z%26sp%3Dr%26sv%3D2018-03-28%26sr%3Db%26rscd%3Dattachment%253B%2520filename%253Deb5a50f9e98f50b9_2022-01-28--10-48-41--0--qlog.bz2%26sig%3DBbgjHIvr7quLCnusOedMAK8mbhTgs2%2Fn0MKXkNudPS8%253D&event=carParams

Or click on the route in useradmin, then on the first segment. In the drop-down click `carParams` and you can see if it was using openpilot longitudinal control or not.

tecandrew, 2022-02-06:
eb5a50f9e98f50b9|2022-02-06--09-36-58--0

Here's a short snippet with cruise toggling on/off by the cruise button, cancel button, and brake with OP disabled.

sshane, 2022-02-11:
It looks like it may be a possibility that openpilot is sending cancel button messages with a counter value that's confusing one of the ECUs. You can see here when openpilot sends the cancel button message, its internal frame counter value doesn't match the car's: https://user-images.githubusercontent.com/25857203/153536894-0ef24e00-f8d4-4f4d-a0ed-98e7d7cfe87e.png

I'll make a branch where openpilot copies the current counter value + 1 and have you try it to see if it disengages on gas all the time.

sshane, 2022-02-12:
So we actually discovered that in some new Hyundais the cancel button also means resume, and so the ECU that controls the engaged status receives a multiple cancel button messages from openpilot, causing a quick disengage and then engage. So we need to find some other way of cancelling ACC that doesn't use a signal that means both engage and disengage, based on the current engaged status.

Can you confirm that your main on button (usually the button that turns the cruise control functionality on or off) also can enable ACC?

tecandrew, 2022-02-12:
some photos to help. the main cruise control button (1) enables the ACC/SCC, yes. there is also the resume button (2) that can engage/disengage ACC/SCC if the cruise control is toggled on. the resume button is also used as an alternative to the gas pedal to reengage while in traffic when the car has come to a complete stop 🤔 (photo from the owner's manual)

as a side note: the latter note is also frequently brought up in #hyundai-kia-genesis as a bug while OP is enabled, because some newer HKGs constantly see screen on their dash and hear the frequent "ding" alert when the car has come to a complete stop

[image]

[image]

tecandrew, 2022-03-02:
`eb5a50f9e98f50b9|2022-03-02--08-14-35--0`

full drive uploaded. the car threw some _really_ interesting errors with this branch during my ~8 min test. OP disengaging on gas appears to be isolated and cancels as it should, but now shows seemingly erratic behavior in the process now:

- Door Open warning on the cluster appears everytime gas is used to disengage
  - the Hyundai FCA sound sound is very audible
- OP disables -> goes back to stock SCC/ACC feature that allows gas to be pressed to resume
- OP disable by gas w/ lead car slowing down -> erratic FCA collision warnings occur

[image]

https://user-images.githubusercontent.com/21319730/156408157-38ea063c-5a51-48ca-9e46-e947c9a2e350.mp4

https://user-images.githubusercontent.com/21319730/156408153-07d17bdd-8061-4baf-9893-6eb42a75dbc8.mp4

sshane, 2022-03-03:
Thanks for the videos! In the second one, it looks like sending the door open bit continuously didn't cancel cruise at all, which is strange, but means we can't use the method.
[image]

sshane, 2022-03-09:
This is our 2020 Sonata on the branch you tried vs. the more reliable branch I came up with using the same door open bit:
[image]

And this is the unfiltered distplot of the old branch you tried on our car (>50 frames):
[image]

closed 2022-04-26 by sshane