commaai/openpilot #23695: PC: Fix model process when using CAN replay

sshane, 2022-02-02:
### Describe the bug

When using CAN replay on PC with jungle and panda, modeld crashes with:

```
Traceback (most recent call last):
  File "/home/batman/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 60, in <module>
    run_loop(ort_session)
  File "/home/batman/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 34, in run_loop
    inputs.append(read(ts).reshape(shp))
  File "/home/batman/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 17, in read
    assert(len(st) > 0)
AssertionError
```

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

NA

### openpilot version

NA

adeebshihadeh, 2022-02-02:
probably a duplicate of  #23666

closed 2022-02-02 by sshane

maykonpacheco, 2022-02-03:
that's right