commaai/openpilot #23705: athenad: ignore duplicate upload requests

gregjhogan, 2022-02-04:
If two requests come in to upload the same file to the same URL (excluding query params) the second should be ignored if the first is still in the queue or is currently uploading.

pd0wm, 2022-02-08:
Doesn't it make more sense to remove the existing one and replace it by the new one?

gregjhogan, 2022-02-08:
That would be better.  I was worried there might be race conditions related that would result in it still getting uploaded twice.  Now that I think about it again, maybe if a request to upload the same file to the same host name comes in we can update the URL in the existing queue item to accomplish this?

zutila, 2024-02-20:
@adeebshihadeh

gregjhogan, 2024-02-24:
This was closed by #25062

closed 2024-02-24 by gregjhogan