commaai/openpilot #23706: Toyota: use common brake pressed signal

sshane, 2022-02-04:
### Describe the bug

This PR was to use a bit now on a message called `BRAKE_2`, https://github.com/commaai/openpilot/pull/19638

However there exists another bit on `PCM_CRUISE_2` that seems to be common across all platforms (need to check nodsu): https://github.com/commaai/opendbc/pull/551

If it is, and it matches the `BRAKE_2` message on the Lexus ES, then we can use that bit in openpilot and panda.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Toyota

### Provide a route where the issue occurs

NA

### openpilot version

NA

sshane, 2022-04-22:
It appears this signal is sometimes 1 when the normal brake pressed bit is 0 when when there's a build-up of brake pressure that the ACC system (nor the user) isn't requesting. Might be able to check for mismatches for the Sienna issue here: #24146

sshane, 2023-08-07:
Unclear if this is feasible and not needed at the moment.

adeebshihadeh, 2023-09-29:
@sshane still valuable to keep this open?