commaai/openpilot #23709: Comma 3 crashed/rebooted in the middle of a drive

brinew27, 2022-02-04:
### Describe the bug

This is the first time this has happened to me on either the comma 2 or comma 3, but my comma appeared to freeze/crash in the middle of a drive. I was driving with OpenPilot engaged, I first heard a "ding" from my car essentially that the Comma 3 had stopped handling driving functions/disconnected, then the Comma3 made a loud beeping noise, even though the image of my drive on the device was frozen and I had to take control manually as I was drifting out of lane. The device then rebooted. The crash happened at the end of the attached route, and then the next route (286c142afbb0252e|2022-02-03--15-58-24) was after the device rebooted. 

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

286c142afbb0252e|2022-02-03--15-47-21

### openpilot version

0.8.12

pd0wm, 2022-02-04:
Looks like kernel panic somewhere in the wifi stack. Will look into it!

Relevant logs:
```
<3>[  639.545372] Fatal error on the modem.
<3>[  639.545432] modem subsystem failure reason: modem_xpu_int.c:456:XPU violation End.
<6>[  639.545441] subsys-restart: subsystem_restart_dev(): Restart sequence requested for modem, restart_level = RELATED.
<6>[  639.550219] ipa-wan ipa3_ssr_notifier_cb:2692 IPA received MPSS BEFORE_SHUTDOWN
<3>[  639.554331] ipa client_lock_unlock_cb:692 Invalid EP
<3>[  639.554374] ipa client_lock_unlock_cb:692 Invalid EP
<6>[  639.554482] rmnet_ipa started deinitialization
<3>[  639.554496] ipa ipa3_teardown_sys_pipe:1113 bad parm.
<3>[  639.554519] ipa-wan ipa3_wwan_remove:2516 Failed to teardown IPA->APPS pipe
<3>[  639.554542] ipa ipa3_teardown_sys_pipe:1113 bad parm.
<3>[  639.554563] ipa-wan ipa3_wwan_remove:2521 Failed to teardown APPS->IPA pipe
<3>[  639.555265] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister rmnet_ipa0
<3>[  639.644104] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister rmnet_ipa0
<3>[  639.644311] ipa-wan ipa3_del_dflt_wan_rt_tables:413 Deleting Route hdl:(0x7) with ip type: 0
<3>[  639.646792] ipa-wan ipa3_del_dflt_wan_rt_tables:422 Deleting Route hdl:(0x9) with ip type: 1
<6>[  639.648250] rmnet_ipa completed deinitialization
<6>[  639.648700] ipa-wan ipa3_ssr_notifier_cb:2708 IPA BEFORE_SHUTDOWN handling is complete
<6>[  639.648737] icnss: PD service down, pd_state: 2, state: 0x4f8a: cause: Root PD crashed
<3>[  639.649362] icnss: PD Down while recovery inprogress, crashed: 1, state: 0x4f8a
<3>[  639.649408] icnss: ASSERT at line 2472
<4>[  639.649448] ------------[ cut here ]------------
<2>[  639.649458] Kernel BUG at ffffff94c198e894 [verbose debug info unavailable]
<4>[  639.649484] ------------[ cut here ]------------
<2>[  639.649497] Kernel BUG at ffffff94c198e894 [verbose debug info unavailable]
<0>[  639.649512] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
<4>[  639.649528] Modules linked in: wlan(CE) snd_soc_sdm845(E) snd_soc_wcd9xxx(E)
<4>[  639.649568] CPU: 1 PID: 6 Comm: kworker/u16:0 Tainted: G         C  E   4.9.103+ #26
<4>[  639.649581] Hardware name: Qualcomm Technologies, Inc. sda845 v2.1 TurboX-SOM_V01 (DT)
<4>[  639.649607] Workqueue: icnss_driver_event icnss_driver_event_work
<4>[  639.649628] task: ffffffe1b528c600 task.stack: ffffffe1b5388000
<4>[  639.649643] PC is at icnss_driver_event_work+0x131c/0x18d0
<4>[  639.649657] LR is at icnss_driver_event_work+0x143c/0x18d0
<4>[  639.649671] pc : [<ffffff94c198e894>] lr : [<ffffff94c198e9b4>] pstate: 40c00145
<4>[  639.649683] sp : ffffffe1b538bc60
<4>[  639.649696] x29: ffffffe1b538bc60 x28: ffffff94c43a6000 
<4>[  639.649727] x27: ffffff94c2b844c8 x26: ffffff94c3878000 
<4>[  639.649756] x25: ffffffe112f8a280 x24: dead000000000100 
<4>[  639.649785] x23: ffffff94c26c8870 x22: ffffffe1b2dbf818 
<4>[  639.649814] x21: ffffff94c3a74bf0 x20: ffffff94c43a67b0 
<4>[  639.649844] x19: ffffff94c2b82160 x18: 0000000000000001 
<4>[  639.649873] x17: 0000007fafad4150 x16: 00000000004f9540 
<4>[  639.649902] x15: ffffffffffffffff x14: ffffff94c3878788 
<4>[  639.649931] x13: ffffffe1b538bc21 x12: ffffffe1b538bb50 
<4>[  639.649964] x11: 000000000000098f x10: ffffffe1b538b990 
<4>[  639.649993] x9 : 0000000000003ff0 x8 : 0000000000003fff 
<4>[  639.650022] x7 : ffffffe1bdc56090 x6 : ffffffe1bdc56090 
<4>[  639.650052] x5 : ffffffe1b538ba40 x4 : ffffff94c3879000 
<4>[  639.650082] x3 : ffffff94c3879628 x2 : 054884ef9eae3e00 
<4>[  639.650113] x1 : 0000000000000000 x0 : 0000000000000000 
<4>[  639.650145] 
<4>[  639.650145] PC: 0xffffff94c198e854:
<4>[  639.650159] e854  11000421 b904f101 35000682 f94033e1 90008fb3 91058273 5280aaa2 aa1303e0
<4>[  639.650257] e874  97f1187d f9400680 b40000c0 900092e2 aa1303e1 5280aaa3 91240042 97f094e2
<4>[  639.650354] e894  d4210000 b94097e3 b0008fb6 913ee2d6 b0008c81 aa1603e0 91344021 f90033e1
<4>[  639.650453] e8b4  97f1186d f9400680 b40000c0 295213e3 900092e2 aa1603e1 91240042 97f094d2
<4>[  639.650554] 
<4>[  639.650554] LR: 0xffffff94c198e974:
<4>[  639.650569] e974  97f094a9 39598ac0 35ffa4a0 f94033e1 90008fb3 91058273 52813502 aa1303e0
<4>[  639.650677] e994  97f11835 f9400680 b4fff7c0 aa1303e1 900092e2 52813503 91240042 97f0949a
<4>[  639.650779] e9b4  17ffffb8 36ff8ac0 17ffff41 f9414f23 d0008fbb 9110437b b0008c81 aa1b03e0
<4>[  639.650876] e9d4  91344021 2a1603e2 f90033e1 97f11822 f9400680 b40000e0 f9414f24 900092e2
<4>[  639.650971] 
<4>[  639.650971] SP: 0xffffffe1b538bc20:
<4>[  639.650984] bc20  c198e9b4 ffffff94 b538bc60 ffffffe1 c198e894 ffffff94 40c00145 00000000
<4>[  639.651079] bc40  c3879000 ffffff94 ffffffff 00ffffff ffffffff ffffffff bdc56090 ffffffe1
<4>[  639.651180] bc60  b538bd70 ffffffe1 c14cb6bc ffffff94 b2dbfa50 ffffffe1 b5236500 ffffffe1
[... 84 lines cut ...]
<4>[  639.672420] [<ffffff94c14b2078>] do_exit+0x6d8/0x9b0
<4>[  639.672433] [<ffffff94c148bc78>] die+0x240/0x258
<4>[  639.672443] [<ffffff94c148bcdc>] bug_handler.part.2+0x4c/0x78
<4>[  639.672449] [<ffffff94c148bd44>] bug_handler+0x3c/0x48
<4>[  639.672459] [<ffffff94c1485204>] brk_handler+0x9c/0xd0
<4>[  639.672468] [<ffffff94c1481874>] do_debug_exception+0xa4/0x170
<4>[  639.672474] Exception stack(0xffffffe1b538ba60 to 0xffffffe1b538bb90)
<4>[  639.672481] ba60: ffffff94c2b82160 0000007fffffffff ffffffe1b538bc60 ffffff94c198e894
<4>[  639.672489] ba80: 0000000040c00145 000000000000003d ffffffe112f8a280 ffffffe1b538bc60
[... cut ...]
<4>[  639.672545] bb80: ffffffe1b538b990 000000000000098f
<4>[  639.672553] [<ffffff94c1483660>] el1_dbg+0x18/0x74
<4>[  639.672560] [<ffffff94c14cb6bc>] process_one_work+0x20c/0x4d8
<4>[  639.672567] [<ffffff94c14cb9d8>] worker_thread+0x50/0x4d0
<4>[  639.672576] [<ffffff94c14d2980>] kthread+0x100/0x108
<4>[  639.672582] [<ffffff94c1483f00>] ret_from_fork+0x10/0x50
<0>[  639.672590] Code: d503201f f942e660 f9400bf3 a8c27bfd (f85d8000) 
<4>[  639.672600] ---[ end trace 3cc23375261c2038 ]---
```

redacid95, 2022-02-04:
Could this have been fixed with https://github.com/commaai/openpilot/pull/23701? @pd0wm

pd0wm, 2022-02-07:
No, this is a kernel issue

adeebshihadeh, 2024-01-05:
Haven't seen this in a while

closed 2024-01-05 by adeebshihadeh