commaai/openpilot #23728: Car unrecognized after upgrade to v0.8.12 - Corolla Hybrid 2021

CPontes, 2022-02-08:
### Describe the bug

After I upgraded to version 0.8.12 my Corolla hybrid 2021, doesn’t recognizes the car. I have reinstalled the software but still didn’t work. It was working fine on version 0.8.11 

### What hardware does this issue affect?

comma two

### Which car does this affect?

Toyota Corolla Hybrid 2021 

### Provide a route where the issue occurs

cb7a33a4e4f4116b|2022-02-07--15-41-54--0

### openpilot version

0.8.12

CPontes, 2022-02-08:
Thank you so much for your fast reply and help. Yes, all the cables are connected and well plugged. I also tried to disconnect and connect again all of them. The panda shows connected on the comma device. But it doesn't recognizes the car. Is there a way to install version 0.8.11 to check if it will work again?

pd0wm, 2022-02-08:
As far as I can tell this is a hardware issue and downgrading won't help. Are you sure the ethernet cables are plugged in correctly? Maybe try with a different ethernet cable.

CPontes, 2022-02-08:
I changed the cable and it worked! Thank you so much 😃👍🏻👏🏻

closed 2022-02-08 by CPontes