commaai/openpilot #23736: Bolt EV

Sid928, 2022-02-09:
I would really like to get the Comma ai system set up on my 2020 Bolt EV.  Can anyone help?

adeebshihadeh, 2022-02-09:
Car support largely comes from the community. Check out our [Discord](https://discord.comma.ai) or GitHub Discussions for getting started.

closed 2022-02-09 by adeebshihadeh

Sid928, 2022-02-09:
I posted a message.  I hope someone will be able to help.

Sid928, 2022-02-09:
As you may know, the recalled Bolts are now getting battery packs to replace the recalled ones.  I hope that you will be able to get the years of the Bolts updated and supported at some point.

Thanks,
Sid

Closed #23736<https://github.com/commaai/openpilot/issues/23736>.

darknight111, 2022-02-09:
Hi Sid,

Check out the opgm repository on GitHub. They is the most up to date Bolt and GM development.