commaai/openpilot #23778: 2019 Rav4 takes forever to pull away from stop, to the point where unusable

zorrobyte, 2022-02-16:
### Describe the bug

Stop and go perf is unusable as it takes forever for the car to accelerate when the lead starts moving. This leads to cars honking behind me and getting passed by angry drivers. Stoplights are not chill.

### What hardware does this issue affect?

comma three

### Which car does this affect?

2019 Toyota Rav 4

### Provide a route where the issue occurs

4aa5d874f367adbc|2022-02-10--09-55-16--8

### openpilot version

9535dc606dddfe6f0807e0b7b2e21867304e507a

Verylukyguy, 2022-02-16:
My GMC Acadia is the same, it barely creeps away from a stop.  Using a fork that Disables Disengagement on Gas helps temporarily circumvent this problem until a fix is developed.

sshane, 2022-02-17:
I can't speak to the planning and MPC side of things, but it looks like the stock ACC system wanted to start moving when *our* vTargetFuture reached 0.2 m/s, while openpilot waited until it was 0.5 m/s (a difference of 0.56 seconds later here). I'll add this as data to my PR here: https://github.com/commaai/openpilot/pull/23426

This won't fix everything however, as our acceleration request ramps down after we start moving because P ramps down as we approach the desired speed, and Toyota is really sensitive to changes in acceleration sometimes:
[image]

HaraldSchafer, 2022-03-03:
We need some better logic to deal with the low-speed behavior in most cars, they tend to be very slow to react to accel requests from stationary. It's on our list of TODOs!

closed 2023-09-02 by adeebshihadeh