commaai/openpilot #23797: Openpilot unavailable Waiting for controls to start

Reynhornwood, 2022-02-19:
### Describe the bug

I have tried rebooting, hard reset, reinstalling software, checking all chords are snug, and tried 2 different UBC C chords. Nothing seems to work. 

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

76 and college

### openpilot version

0.8.12

DaVinciCaesar, 2022-02-20:
> This means the device isn't getting any CAN data. Checkout discord for
> more help.

Guy and discord novice [openpilot] How do I get trained ?

I am new to discord. How do I check it?
I feel like a total noob. I just need help


-- 
____________________________________
Reyn Hornwood
Cell: 310.420.0077
BRE License: CA 01955792

DaVinciCaesar, 2022-02-20:
Git* sorry, I'm a total noob



-- 
____________________________________
Reyn Hornwood
Cell: 310.420.0077
BRE License: CA 01955792

Reynhornwood, 2022-02-21:
I actually needed a new USB-C cable. I ordered a new one on Amazon, and
plugged it in. works great. I am so gratefdu
____________________________________
*Reyn Hornwood*
*Cell: **310.420.0077*
<3104200077>
BRE: 01955792

zeroyang0809, 2023-02-20:
hello i meet the same question
do you fix this problem
could you share some steps to fix it 
thx