commaai/openpilot #23804: Bigmodel: Phantom braking, possible incorrect lead

zorrobyte, 2022-02-20:
### Describe the bug

[image]

The lead indicator was tracking the car on the exit ramp right and OP was braking, even though I had nothing in front of me.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

4aa5d874f367adbc|2022-02-19--19-45-14--10

### openpilot version

85efde269da2680a77c75c248aafc8c29435ca98

### Additional info

I replicated this previously, and also had issues in another instance stopping for a lead. See https://github.com/commaai/openpilot/pull/23684#issuecomment-1042371661 and following

zorrobyte, 2022-02-20:
[image]

vanstinator, 2022-02-20:
Comma 2, same behavior, same sha (https://github.com/commaai/openpilot/commit/85efde269da2680a77c75c248aafc8c29435ca98). I was in the right exit lane and there was a car in front of me and a car in the middle lane a little further ahead. Lead indicator jumped between 2 vehicles while I was in an exit.

`27b1608e5d22ba94|2022-02-19--20-24-05--16`

[image]

zorrobyte, 2022-02-20:
This stop felt pretty rough and reminds me of controls lag from previously, could have just been expected behavior, but still wanted to include:
`4aa5d874f367adbc|2022-02-19--23-05-39--19`

I have a route from a previous commit that had the same phantom braking behavior around the same place on the interstate:
`4aa5d874f367adbc|2022-02-16--14-26-49`

Here is a disengagement due to very laggy controls, as Harold mentioned in the commit:
`4aa5d874f367adbc|2022-02-16--18-50-12--5`

[image]

scottadair, 2022-02-20:
Had something similar happen to me. Driving on the highway and came around a corner which had an on-ramp. OP locked onto the cars on the merge and slowed down.

4b8ae689eaa74602|2022-02-20--11-30-16

JoshHobbs, 2022-02-21:
Here's one where I believe it was braking for oncoming traffic.

255403b2bd0acb75|2022-02-21--17-32-13--4

HaraldSchafer, 2022-02-22:
There was a bug in how the bigmodel wideframe was transformed. Which had a pretty big impact on performance. This should fix
https://github.com/commaai/openpilot/pull/23817

@zorrobyte Btw that laggy control disengagement has nothing to do with bigmodel, the phantom braking though is probably this bug

zorrobyte, 2022-02-22:
> There was a bug in how the bigmodel wideframe was transformed. Which had a pretty big impact on performance. This should fix #23817
> 
> @zorrobyte Btw that laggy control disengagement has nothing to do with bigmodel, the phantom braking though is probably this bug

Thank you for looking into it and the fix so quickly! I'll keep an eye out for the new model and look forward to testing it, will be happy to close this issue once resolved

closed 2022-02-22 by HaraldSchafer