commaai/openpilot #23809: Toyota TSS 2.5: openpilot can learn bad offset if initialized in a turn

sshane, 2022-02-21:
### Describe the bug

Something I noticed on my TSS 2.5 Camry Hybrid is that `STEER_TORQUE_SENSOR`'s angle signal is absolute and not relative, meaning it doesn't start at 0 and very closely follows the angle in `STEER_ANGLE_SENSOR` without any offset (unlike TSS2 where `STEER_TORQUE_SENSOR`'s angle is relative.

Meaning if openpilot goes onroad while the car is driving and we're in a turn, a bad offset can be learned. This isn't a problem with torque control, as the desired angle is also offset in this way, however, when testing the LTA message which uses desired angles, the issue is very apparent as the car tried to veer left multiple times after engagement.

https://imgur.com/a/s31NKlQ

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

f15e3c37c118e841|2022-02-20--16-53-06--0

### openpilot version

0.8.14

pd0wm, 2022-02-21:
Is there a another signal/bit to indicate the measurement can be used?

pd0wm, 2022-02-21:
Seems like bit 3 in the first byte of the `STEER_TORQUE_SENSOR` messages is one when initializing.