commaai/openpilot #23812: CAN Errors after 0.8.13 Update: Comma 2 & 2022 Camry Hybrid

JamesL623, 2022-02-21:
### Describe the bug

I have had multiple CAN errors after updating to 0.8.13.

Dongle ID: 69567f65ae707320

Here is a drive with a CAN error: 69567f65ae707320|2022-02-20--19-21-49--0

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

69567f65ae707320|2022-02-20--19-21-49--0

### openpilot version

0.8.13

connorlarkin1, 2022-02-22:
I have also been getting intermittent CAN errors after updating to `0.8.13` in my 2021 RAV4H (comma 2). It happens intermittently and usually resolves itself after restarting the car. I swapped out my OBD-C cable and checked for stock passthrough to ensure it wasn't a harness issue. 

Route with the CAN Error:

`5ab891a2663f10be|2022-02-19--21-29-42--0` 

Route without the CAN Error:

`5ab891a2663f10be|2022-02-19--21-08-55--0`

Additional RAV4 owners reporting the same issue after updating: https://discord.com/channels/469524606043160576/524592892627517450/945574732634402826
https://discord.com/channels/469524606043160576/524327905937850394/944968334179459122

Let me know if I can provide any additional info.

JamesL623, 2022-02-22:
Here is another drive that didn't throw a CAN error, but a communication malfunction error: 
69567f65ae707320|2022-02-21--22-59-23--9

Another drive about 15 minutes after the one above that was throwing a CAN error (I think, could've been comm malfunction). In this same drive, I pull over, restart my car, and OP operates normally and I turn it on a few times during the route:
69567f65ae707320|2022-02-21--23-22-45--0

JamesL623, 2022-02-22:
Here is a normal drive with no errors and OP engaged: 69567f65ae707320|2022-02-22--21-01-29--0

adeebshihadeh, 2022-02-22:
@JamesL623 @connorlarkin1 can you guys try #23825? It seems that there's a not insignificant delay until your gas pedals start sending messages, which makes the EV detection unreliable.

JamesL623, 2022-02-22:
> 

Sure thing. May be able to test tonight, but if not, I will get you some routes tomorrow

adeebshihadeh, 2022-02-23:
Merged into master since the tests passed. We'll push this to the 0.8.13 release once you guys confirm it's reliable now.

JamesL623, 2022-02-23:
**Drive 1:** OP worked perfectly
69567f65ae707320|2022-02-22--23-55-25--0

**Drive 2:** CAN Error
69567f65ae707320|2022-02-23--00-07-28--0

**Drive 3:** OP worked perfectly
69567f65ae707320|2022-02-23--00-08-58--0

Please check to make sure I have the correct changes in my interface and values files. I merged the code over myself and want to be sure there wasn't any errors

adeebshihadeh, 2022-02-23:
@JamesL623 If you're unable to checkout the branch manually, I recommend uninstalling and reinstalling with this URL `installer.comma.ai/commaai/master-ci`. I'm seeing no git diff on those routes.

JamesL623, 2022-02-23:
Darn, forgot to compile after saving the changes to the files.

I’ll get it up and running tomorrow and send the routes to you.

connorlarkin1, 2022-02-23:
The issue is resolved for me. Drove around on `master-ci` and restarted my car a bunch and couldn't reproduce the CAN error message. 

**Routes on `master-ci`:**

`5ab891a2663f10be|2022-02-22--19-26-24`
`5ab891a2663f10be|2022-02-22--19-38-21`
`5ab891a2663f10be|2022-02-22--19-47-51`
`5ab891a2663f10be|2022-02-22--19-51-24`

Thanks @adeebshihadeh for the quick fix!

adeebshihadeh, 2022-02-23:
Thanks for testing @connorlarkin1! We'll get the fixed pushed out to release tomorrow after a bit more time on master.

JamesL623, 2022-02-23:
Just go the update pushed over and will be going for a drive later. Will post routes in here when uploaded

closed 2022-02-23 by adeebshihadeh