commaai/openpilot #23821: Cruise Fault when OP drives below car's stock lower cruise speed limit

tiki-bug, 2022-02-22:
### Describe the bug

Odyssey stock lower speed limit is 22mph.  The following is for an Odyssey without a Comma Pedal.

Expected behavior: when OP is engaged and drives the car below this limit, an info splash tells the user that they are below the limit while OP continues to be able to brake and steer.  Once OP tries to send an acceleration command, OP is disengaged entirely.  OP can be re-engaged once speed is above 25mph.

Current behavior: as soon as car drives below the lower limit while OP is engaged, a Cruise Fault warning comes up forcing the car to be restarted in order to continue using OP.  This also causes dash warnings to be set off, Collision Mitigation and possibly another, until the car is restarted.

The current behavior started after installing 0.8.13

### What hardware does this issue affect?

comma two

### Which car does this affect?

Honda Odyssey 2020

### Provide a route where the issue occurs

0a96f86fcfe35964|2022-02-22--07-40-51

### openpilot version

0.8.13 release2

adeebshihadeh, 2022-02-23:
#23828 has been merged into master and should fix this. Can you test and confirm this fixes your issue? Uninstall openpilot and reinstall with the URL `installer.comma.ai/commaai/master-ci`.

adeebshihadeh, 2022-02-23:
@gricha01 can you test the master branch and confirm the issue is fixed for you? Uninstall openpilot and reinstall with the URL installer.comma.ai/commaai/master-ci.

gricha01, 2022-02-23:
yes that fixed it. i put it in numerous situations and it handled each one perfectly. no faults, no notices, no alarms and came to a complete stop. thanks.

closed 2022-02-23 by adeebshihadeh

tiki-bug, 2022-02-24:
Confirmed this fixed the issue on my Odyssey