commaai/openpilot #23839: Steering bug during turns

jcwillia1, 2022-02-23:
### Describe the bug

during a right turn on my commute on 2/22 5:10-ish pm with open pilot engaged, the car's steering wheel flipped from about 75 degrees to 105 degrees and locked there.  The screen flashed orange to take control.

I disengaged OP (I think, it happened fast) reset the driving line until the lane turned straight and re-engaged cruise.  I initially thought it was a one off issue until someone described the exact same problem on the comma subreddit this morning.

https://connect.comma.ai/587c6380f108936f/1645567568051/1645569124154

17:13 or so

Bonus fun : watch me dodge potholes at 17:08

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

https://connect.comma.ai/587c6380f108936f/1645567568051/1645569124154

### openpilot version

08.13

sshane, 2022-02-23:
It looks like from around 40 degrees you started touching the wheel to override and take the turn
[image]

adeebshihadeh, 2022-02-23:
This is also unrelated to the issue reported on Reddit with the Subaru Ascent.

sshane, 2022-02-24:
Can you verify which segment this occurred? The timestamps seem to be off in connect.

sshane, 2022-02-25:
The take control alert was caused by an internal angle limit in your car, as Hyundai limits you to ~90 degrees and won't steer past that.

[image]

closed 2022-02-25 by sshane