commaai/openpilot #23853: Big Model Laneless going over lines

michaelhonan, 2022-02-25:
### Describe the bug

Seems to randomly go over lines, and seems to have a left-bias, as seen in the linked route. Requires manual intervention to steer back into lane.

Could it be related to the shaking comma two camera from music?

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

4d70bc5e608678be|2022-02-24--12-37-13--3

### openpilot version

c98d9791f049c7cdc0c9154c963755b1239e451c

### Additional info

Video:
https://user-images.githubusercontent.com/12524358/155632318-eb43419d-8617-49d6-8a25-f79bb9224bc1.mp4

redacid95, 2022-03-01:
f1055ddb93ff2d3b|2022-02-27--12-23-08--55

Got a good segment here. C3, Honda Bosch, left drive bias. Gets a bit too close to the left for regular driving, it is super evident when there are construction barriers next to you. 

This is on my master branch which has no major adjustments from the master.

HaraldSchafer, 2022-03-03:
We're doing a big investigation into hugging and non-centered driving on laneless. We hope to have improvements soon.

HaraldSchafer, 2022-03-08:
Related
https://github.com/commaai/openpilot/issues/23922

morrislee, 2022-05-20:
I am driving with laneless model on 8.14, the slower I go, the more it hugs right. If I am going 110km/h, hugging is almost nonexistent. If I switch back to laneful, it is perfectly center straight or curves, so it doesnt seem to be a tuning issue.

I am on a larger vehicle, 2020 GMC Sierra 1500, so any hugging makes me a lot more anxious to take over. Laneful re-centers too much when lanes fades in and out. I am really hoping laneless hugging will be resolved as it holds a straight path really well when lane lines are fading in and out