commaai/openpilot #23870: CAN Error: Check Connections (2020 Lexus NX 300)

sintman, 2022-02-26:
### Describe the bug

Whenever I start the car up, the screen just say's "CAN Error: Check Connections" after openpilot starts.

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

1303439f4af30ab1|2022-02-26--18-32-40

### openpilot version

Release version

### Additional info

Dongle ID: [1303439f4af30ab1]. I have added fingerprinting 2.0 for the car in values.py as per the instructions, the device switches from dashcam to openpilot mode after the change. The CAN Error has been there since I first plugged the device in, before I modified the values.py. I rechecked all the device connections to make them secure but I still see this error.

pd0wm, 2022-02-28:
Has this always been an issue on your Lexus NX? I see you previously used your device successfully on a different car.

sintman, 2022-02-28:
This is my first time trying this on the NX as I switched vehicles. My previous car had no issues and worked

sshane, 2022-03-02:
Hmm, but all your fw versions are identical to everything in the 2018 Lexus NX

sintman, 2022-03-02:
It works! thank you!

closed 2022-03-02 by sintman

sshane, 2022-03-02:
Merged into master!