commaai/openpilot #23904: CAN Error: Check Connections (2022 TOYOTA RAV4 HYBRID)

danielthall, 2022-03-03:
### Describe the bug

I have gone through the steps for fingerprinting my new car, and after modifying the `values.py` the display has gone from "Car Unrecognized: Dashcam Mode" to "CAN Error: Check Connections" after open pilot starts

### What hardware does this issue affect?

comma two

### Which car does this affect?

Toyota RAV4 Hybrid 2022

### Provide a route where the issue occurs

69f9053e1c268efb|2022-03-03--14-31-20

### openpilot version

Release version

### Additional info

Dongle ID: [69f9053e1c268efb]. I think there was a similar issue[similar issue](https://github.com/commaai/openpilot/issues/23870) just closed a few days ago. I have double checked the values in `values.py` but can't see anything immediately wrong.

I have used this device in a 2019 Prius for over a year with no issues.

alancyau, 2022-03-03:
The 2022 RAV4 is currently not an officially OpenPilot supported car. Toyota changed their radar and camera hardware from Denso to Continental for 2022 model year and OP is unable to read the new signals. See discussion here:
https://github.com/commaai/openpilot/discussions/23611

There's currently a workaround to get OpenPilot to steer but the gas/brakes will be handled by the factory ACC system.

danielthall, 2022-03-03:
Well that's unfortunate. Hopefully there is support for the Conti hardware soon. 

In the linked discussion, is the workaround fingerprinting as `CAR.CAMRYH`? If so, are there any guides to fingerprinting as a different vehicle?

alancyau, 2022-03-03:
You can run my branch and see if it recognizes your ecu firmware. There are already a few 2022 users running this branch as a workaround.
https://github.com/alancyau/openpilot/commits/patch-2

When you get to the installer URL, use https://installer.comma.ai/alancyau/patch-2

If you still get car unrecognized, paste your carfw data and I'll add it to the branch.

closed 2022-03-04 by adeebshihadeh

danielthall, 2022-03-04:
That worked great, thanks!