commaai/openpilot #23922: Car hugs right consistently

kinng, 2022-03-07:
### Describe the bug

On lane line mode, my car will always hug the right lane regardless of the calibration and mount position.  The latest laneless mode actually can center it pretty good but gets too close to the lane lines during turns so too scared to use it.  i.e. C3 is capable to drive in the center but somewhere on the lane line mode, it's not working properly.  I have to keep changing the CAMERA_OFFSET value in lane_planner.py after every release.  The current value is -0.13 in release 0.8.13.  The route below has the CAMERA_OFFSET value changed.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Toyota Highlander Hybrid XLE 2017

### Provide a route where the issue occurs

9ff6adaf72343a2c|2022-03-05--10-05-43

### openpilot version

0.8.13

HaraldSchafer, 2022-03-08:
CAMERA_OFFSET doesn't affect the laneless policy. 

Either way, hugging is a problem we are working on. Linking to this issue https://github.com/commaai/openpilot/issues/23853

closed 2022-03-08 by HaraldSchafer

kinng, 2022-03-08:
But the hugging problem is on laneline mode, not landless.

HaraldSchafer, 2022-03-08:
Oh, sorry I thought it was hugging on laneless. I'll check tomorrow!

HaraldSchafer, 2022-03-09:
Hmmm not really sure what the issue is. It's unfortunately not a priority to look into atm, we are fully focused on improving laneless and making laneless the default