commaai/openpilot #23923: Stop and go hesitation

kinng, 2022-03-07:
### Describe the bug

Resume from full stop will hesitate before moving smoothly. i.e. when the lead car stops, my car stops, then when the lead car moves, my car will first release the brake, the car will start to move but I will feel a slight brake being applied, my body will jerk forward a little bit, then the brake is released and the car moves smoothly forward.  Thus every stop and go will cause this unnatural slight body jerk forward before get going.

My car has a C-SDSU installed.  No pedal installed because it's a hybrid.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Toyota Highlander Hybrid XLE 2017

### Provide a route where the issue occurs

9ff6adaf72343a2c|2022-03-05--20-24-40

### openpilot version

0.8.13

j0dan, 2022-11-01:
Same issue here on the 0.8.16 on a 2018 Toyota Sienna.

sshane, 2022-11-01:
Can you post a route with logs uploaded @j0dan?

adeebshihadeh, 2023-09-02:
Closing due to inactivity

closed 2023-09-02 by adeebshihadeh