commaai/openpilot #23926: Snow covered roads, with open tire marks, comma either pulling you off the road or into traffic.

pauleasker, 2022-03-08:
### Describe the bug

On snow covered roads, that have open wheel marks, that is 2 rows of visible road in your lane and 2 rows of visible road in the oncoming lane. Comma wants to pull the vehicle into the furthest to the right open wheel mark in your lane. What results, is the device nearly pulling you off the road. If the left side tire mark is more open, comma wants to pull you and center you over that wheel line. The very dangerous result of this situation is that you are being pulled into oncoming traffic.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

a7f52f253cc2f654|2022-02-24--15-27-32

### openpilot version

0.8.13-release

### Additional info

other routes
a7f52f253cc2f654|2022-02-25--07-43-10
a7f52f253cc2f654|2022-03-07--08-23-05 

[image]

pd0wm, 2022-03-08:
Have you tried using laneless mode (Turn on the "Disable use of lanelines" toggle)?

spektor56, 2022-03-10:
use laneless, it definitely works in the snow.

pauleasker, 2022-03-10:
Ok, I will try it, but I don't know when I will next have snow. Do you run laneless all the time?

spektor56, 2022-03-11:
Yes, I have used laneless exclusively since the alpha release of kale and works fine for me.

HaraldSchafer, 2022-03-18:
Laneless is better at this kind of stuff, laneless will improve over time, these kind of failures will likely never improve for the laneline policy. Laneless still has a couple problems with ping pong and hugging, after that laneless will be the default.

closed 2022-03-18 by HaraldSchafer