commaai/openpilot #23933: I fingerprint carfw to values.py but I got CAN error

iamkimmy80, 2022-03-09:
### Describe the bug

I did check all the wire, flip usb-c but still CAN error showed  

I am in Thailand. 

Thank you. 

### What hardware does this issue affect?

comma two

### Which car does this affect?

Toyota Alphard Hybrid 2021

### Provide a route where the issue occurs

912119ebd02c7a42|2022-03-08--23-31-50

### openpilot version

0.8.13

### Additional info

this is my Toyota Alphard Hybrid 2021 carFw 

carFw': [{'address': 1968, 'ecu': 'esp', 'fwVersion': b'F152658341\x00\x00\x00\x00\x00\x00', 'subAddress': 0},
                         {'address': 1953, 'ecu': 'eps', 'fwVersion': b'8965B58040\x00\x00\x00\x00\x00\x00', 'subAddress': 0},
                         {'address': 2016,
                          'ecu': 'engine',
                          'fwVersion': b'\x0235879000\x00\x00\x00\x00\x00\x00\x00\x00A4701000\x00\x00\x00\x00\x00\x00\x00\x00',
                          'subAddress': 0},
                         {'address': 1872, 'ecu': 'fwdRadar', 'fwVersion': b'\x018821F3301400\x00\x00\x00\x00', 'subAddress': 15},
                         {'address': 1872,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\x028646FV201000\x00\x00\x00\x008646G2601400\x00\x00\x00\x00',
                          'subAddress': 109}],

(Ecu.esp, 0x7B0, None): [b'F152658341\x00\x00\x00\x00\x00\x00'],
(Ecu.eps, 0x7A1, None): [b'8965B58040\x00\x00\x00\x00\x00\x00'],
(Ecu.engine, 0x7E0, None): [b'\x0235879000\x00\x00\x00\x00\x00\x00\x00\x00A4701000\x00\x00\x00\x00\x00\x00\x00\x00'],
(Ecu.fwdRadar, 0x750, 0xF): [b'\x018821F3301400\x00\x00\x00\x00'],
(Ecu.fwdCamera, 0x750, 0x6D): [b'\x028646FV201000\x00\x00\x00\x008646G2601400\x00\x00\x00\x00'],

sshane, 2022-03-19:
Since this is a hybrid, it uses certain different CAN messages, like the gas pedal. I'll make a branch for you

sshane, 2022-03-19:
Can you try the branch on this PR out? https://github.com/commaai/openpilot/pull/24003

We also need a test route, we just need a short drive from point A to point B where you go from parking lot to parking lot, as it will be public so all forks can run tests on the route as well

iamkimmy80, 2022-03-19:
Thank you very much. please tell me how to install the branch.

sshane, 2022-03-19:
You can type https://installer.comma.ai/commaai/alphardh-tss2 in the custom software dialog box

iamkimmy80, 2022-03-19:
it work. i've done my first drive, a little bit longer than A to B, got so excited. what do i have to do next?

adeebshihadeh, 2022-03-19:
@iamkimmy80 great, post the route here, so we can use it for regression tests. Once you post the route, we can merge the PR, and your car will be fully supported in the next release. The route being longer is fine, but the whole thing will be public.

iamkimmy80, 2022-03-19:
912119ebd02c7a42|2022-03-19--07-24-50

is it right? 

thank you.

closed 2022-03-20 by adeebshihadeh