commaai/openpilot #23935: Lateral MPC is undershooting future predicted plan

sshane, 2022-03-09:
It seems the lateral MPC's plan is much not very when compared to the actual future plan in `steerActuatorDelay` seconds. We can't interpolate the lateral plan until this is fixed, but I'll look into it some more and confirm it's a real issue. Closing https://github.com/commaai/openpilot/pull/23788 until it doesn't cause oscillations.

Longitudinal:
[image]

Lateral:
[image]

GregorKikelj, 2022-03-10:
Do you take this into account? https://github.com/commaai/openpilot/blob/4f4f18a648f8b5ec1fa63c60f69da1a32e60d9a5/selfdrive/controls/lib/drive_helpers.py#L99

sshane, 2022-03-10:
Yup, I compared now + 0.2 seconds vs. the actual plan in 0.2 seconds

ClockeNessMnstr, 2022-04-06:
Is the orange line time  shifted curvatures[0] and the blue line OP's current get_desired_curvature? For 0.+.2s

Was about to mention they looked right but this would be backwards from expected I think

GregorKikelj, 2022-04-06:
@ClockeNessMnstr I think it's exactly that yeah. Got better match here by using 
`psi_interp = interp(0.2, T_IDXS[:CONTROL_N], lat_plan.psis)
psi_interp_curvature = psi_interp/(max(CS.vEgo, 0.001)*0.2)`

adeebshihadeh, 2022-08-14:
Have you confirmed this is a real issue?

sshane, 2022-08-17:
Well I assume so since we can't interpolate the plan to 100Hz. However I haven't confirmed that it's an issue *recently*, so if anyone doesn't do that, I'll check in the next few weeks

adeebshihadeh, 2023-06-24:
@sshane is this still relevant?

sshane, 2023-06-25:
Yassine was going to check it out

adeebshihadeh, 2023-08-20:
Closing, @YassineYousfi feel free to re-open if it's still an issue

closed 2023-08-20 by adeebshihadeh