commaai/openpilot #23944: Toyota: BSM detection is flaky

sshane, 2022-03-10:
In route `4289612fe1127f18|2022-03-10--04-13-46` (and probably many others), we miss the 1 Hz BSM message and openpilot thinks there's no BSM sensors, when in actuality we just miss the low frequency message.

adeebshihadeh, 2022-03-10:
Let's see if there's a higher frequency message that also indicates whether there's BSM. I'd like to avoid increasing the min fingerprinting time.

sshane, 2022-03-10:
Kumar on Discord said there's another message which sends at 5 Hz:

[image]

adeebshihadeh, 2022-03-10:
See https://github.com/commaai/opendbc/pull/190

sshane, 2022-07-07:
Fixed in https://github.com/commaai/openpilot/pull/24964

closed 2022-07-07 by sshane