commaai/openpilot #23963: communicate when driver monitoring is calibrating

gregjhogan, 2022-03-14:
It is a bad user experience when you constantly get alerts saying you are not paying attention (and falling back to wheel touch) due to camera based driver monitoring not working because it is calibrating (nothing communicates to the user what is happening)

ZwX1616, 2022-03-14:
It still uses the camera policy, just the default offsets instead of the learned ones. Either way it should not say you're distracted when paying attention. What we really should fix is the false positives from the model. But I agree that it's confusing to have constant alerts then it's fine after a while.

ClockeNessMnstr, 2022-04-06:
On the C3 I had to raise the dm box/window and reverse the sign on "natural pitch"

My head was cropped out partially at the top of the box. Not sure if having my face higher than normal produces a "rotation" in the fisheye view.

This "calibration" stage seemed ok and stopped giving me issues after the pitch change but my changes were just a quick fix I had to do before a road trip and haven't bothered to revert/test. 

This sounds like the issue I had though so figured I'd mention it again

adeebshihadeh, 2022-10-13:
@ZwX1616 is this still an issue?

ZwX1616, 2022-11-04:
uncalibrated FPs should be much improved with #26348

closed 2022-11-04 by ZwX1616