commaai/openpilot #23984: 2022 Mazda CX-5 - Steer Unavailable Below 0 mph

alancyau, 2022-03-17:
### Describe the bug

When OpenPilot is engaged, the `Steer Unavailable Below 0 mph` message shows up on screen. I verified the car steers down to 0.

[image]

### What hardware does this issue affect?

comma three

### Which car does this affect?

2022 Mazda CX-5

### Provide a route where the issue occurs

N/A

### openpilot version

0.8.13

sshane, 2022-03-19:
@alancyau care to test this? https://github.com/commaai/openpilot/pull/23998

alancyau, 2022-03-19:
> @alancyau care to test this? #23998

I’ll test this tomorrow and let you know!

alancyau, 2022-03-19:
@sshane I can confirm this has resolved the issue, thanks!

alancyau, 2022-06-15:
Can we reopen this issue? I'm getting `Steer Unavailable below 0 mph` when the car comes to a complete stop while OP is engaged. Running on 0.8.14 C3.

I'm not sure if this is related to this current issue, but the car doesn't appear to automatically accelerate when the lead car moves again. I had to override and manually step on the gas pedal since there was a car behind me and the lead car was pretty far ahead. Car uses stock long. 

a4af1602d8e668ac|2022-06-15--15-00-14--2

adeebshihadeh, 2022-06-15:
@sshane can you look into this?

sshane, 2022-06-15:
The frame he hits 0 mph is when LKAS_BLOCk rises, so looks like the EPS doesn't steer at true 0.

[image]

mallorbc, 2023-05-10:
I have noticed this issue on the latest main release for 2022 Mazda Cx5 as well.

alancyau, 2023-11-23:
Not sure if this is still a bug on the latest release as I've been without OP for a while. I'll have a C3 to test in a few weeks to verify if this issue still exists.

closed 2023-12-18 by alancyau